botletics/SIM7000-LTE-Shield #200: HTTPS refused on SIM7070G

spara7C5, 2020-09-29:
Hi, the http connection and get/post method work well. But https don`t.
I downloaded the certificare using google chrome.
Here the AT sequence command:
Modem Initializing starts:
AT command:  AT
AT command:  ATE1V1
AT response: ATE1V1
OK
AT command:  ATI
AT response: ATI
R1951.01

OK
AT command:  AT+CGMR
AT response: AT+CGMR
Revision:1951B03SIM7070

OK
AT command:  AT+CMEE=2
AT response: AT+CMEE=2
OK
AT command:  AT+CGNAPN
AT response: AT+CGNAPN
+CGNAPN: 0,""

OK
AT command:  AT+CFUN=0
AT response: AT+CFUN=0
OK
AT command:  AT+CGDCONT=1,"IP","ibox.tim.it"
AT response: AT+CGDCONT=1,"IP","ibox.tim.it"
OK
AT command:  AT+CFUN=1
AT response: AT+CFUN=1
OK
AT command:  AT+CGATT?
AT response: AT+CGATT?
+CGATT: 0

OK
AT command:  AT+CSQ
AT response: AT+CSQ
+CSQ: 14,99

OK
AT command:  AT+CGREG?

+CGREG: 0,2

OK
AT command:  AT+CGATT?
AT response: AT+CGATT?
+CGATT: 0

OK
AT command:  AT+CSQ

+CSQ: 14,99

OK
AT command:  AT+CGREG?

+CGREG: 0,2

OK
AT command:  AT+CGATT?
AT response: AT+CGATT?
+CGATT: 1

OK
AT command:  AT+CSQ

+CSQ: 17,99

OK
AT command:  AT+CGREG?

+CGREG: 0,1

OK
AT command:  AT+CGATT?
AT response: AT+CGATT?
+CGATT: 1

OK
AT command:  AT+CGNAPN
AT response: AT+CGNAPN
+CGNAPN: 0,""

OK
AT command:  AT+CNCFG=0,1,"ibox.tim.it"
AT command:  AT+CNACT=0,1
AT response: AT+CNACT=0,1
OK
AT PLUS MESSAGE: 

+APP PDP: 0,ACTIVE
AT command:  AT+CNACT?
AT response: AT+CNACT?
+CNACT: 0,1,"10.67.251.234"
+CNACT: 1,0,"0.0.0.0"
+CNACT: 2,0,"0.0.0.0"
+CNACT: 3,0,"0.0.0.0"

OK
AT command:  AT+CFSGFRS?

+CFSGFRS: 5888000

OK
AT command:  AT+CFSINIT
AT response: AT+CFSINIT
OK
AT command:  AT+CFSWFILE=?
AT response: AT+CFSWFILE=?
+CFSWFILE: (0-3),50,(0-1),(1-10240),(100-10000)

OK
AT command:  AT+CFSWFILE=3,"httpbin_root_ca.crt",0,1943,10000
AT response: AT+CFSWFILE=3,"httpbin_root_ca.crt",0,1943,10000
DOWNLOAD
AT response: 
OK
AT command:  AT+CFSTERM
AT response: AT+CFSTERM
OK
AT command:  AT+CSSLCFG="convert",2,"httpbin_root_ca.crt"
AT response: AT+CSSLCFG="convert",2,"httpbin_root_ca.crt"
OK
AT command:  AT+SHSSL=1,"httpbin_root_ca.crt"
AT response: AT+SHSSL=1,"httpbin_root_ca.crt"
OK
AT command:  AT+SHCONF="URL","https://www.httpbin.org"
AT response: AT+SHCONF="URL","https://www.httpbin.org"
OK
AT command:  AT+SHCONF="BODYLEN",1024
AT response: AT+SHCONF="BODYLEN",1024
OK
AT command:  AT+SHCONF="HEADERLEN",350
AT response: AT+SHCONF="HEADERLEN",350
OK
AT command:  AT+SHCONF="IPVER",0
AT response: AT+SHCONF="IPVER",0
OK
AT command:  AT+SHCONF="TIMEOUT",30
AT response: AT+SHCONF="TIMEOUT",30
OK
AT command:  AT+SHCONN

+CME ERROR: operation not allowed

SHCONN fails as you can see...
Can you help me?

spara7C5, 2020-09-29:
this is the certicate FIY:

-----BEGIN CERTIFICATE-----\n\
MIIFbjCCBFagAwIBAgIQC6tW9S/J9yHIw1v8WOnMPDANBgkqhkiG9w0BAQsFADBG\n\
MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRUwEwYDVQQLEwxTZXJ2ZXIg\n\
Q0EgMUIxDzANBgNVBAMTBkFtYXpvbjAeFw0yMDAxMTgwMDAwMDBaFw0yMTAyMTgx\n\
MjAwMDBaMBYxFDASBgNVBAMTC2h0dHBiaW4ub3JnMIIBIjANBgkqhkiG9w0BAQEF\n\
AAOCAQ8AMIIBCgKCAQEApFxnGvqYGUel320/nRE281GA6WAOVwY+Npl79AIz45bH\n\
XcxNu+LeMEuGBvrl2EuccQJGXpCY8+sCzFRmcCZsMtTzUdj6R/QbWR7OFjf6Z6w1\n\
AiKccc7iKlRUF/tWAuoLr1b6L9+JfAkJAUL35VV7/vIs9IZ8uWJDhEB2wU6rRZO+\n\
2RBvHGM7oeBNda1/maukjLNYmJ+pxSnrsRTMh3dHUCxZ47h2UZhj2SWCPlW+SMsY\n\
NM/JkURnzSy0lgq/woVeM5g4nOpWuljO1scJ0ZRbR3I+3JveGEd3sQi8e6HkWFtI\n\
mGklhirXxE/t86GP86s+XbwnABIML12h09M3mTJqEwIDAQABo4IChjCCAoIwHwYD\n\
VR0jBBgwFoAUWaRmBlKge5WSPKOUByeWdFv5PdAwHQYDVR0OBBYEFE1H1xvaOuX7\n\
0DFAys411lS5yO+lMCUGA1UdEQQeMByCC2h0dHBiaW4ub3Jngg0qLmh0dHBiaW4u\n\
b3JnMA4GA1UdDwEB/wQEAwIFoDAdBgNVHSUEFjAUBggrBgEFBQcDAQYIKwYBBQUH\n\
AwIwOwYDVR0fBDQwMjAwoC6gLIYqaHR0cDovL2NybC5zY2ExYi5hbWF6b250cnVz\n\
dC5jb20vc2NhMWIuY3JsMCAGA1UdIAQZMBcwCwYJYIZIAYb9bAECMAgGBmeBDAEC\n\
ATB1BggrBgEFBQcBAQRpMGcwLQYIKwYBBQUHMAGGIWh0dHA6Ly9vY3NwLnNjYTFi\n\
LmFtYXpvbnRydXN0LmNvbTA2BggrBgEFBQcwAoYqaHR0cDovL2NydC5zY2ExYi5h\n\
bWF6b250cnVzdC5jb20vc2NhMWIuY3J0MAwGA1UdEwEB/wQCMAAwggEEBgorBgEE\n\
AdZ5AgQCBIH1BIHyAPAAdgDuS723dc5guuFCaR+r4Z5mow9+X7By2IMAxHuJeqj9\n\
ywAAAW+2QhJbAAAEAwBHMEUCIQCBAIJ4tACBrdHwB4ZnGIGTy3/9FxuZ9GIoHgfX\n\
5RjefQIgeXY+x7oWQmIShXCrBSdqeTYXrsxQcWE6ZpAyQxcsdUUAdgCHdb/nWXz4\n\
jEOZX73zbv9WjUdWNv9KtWDBtOr/XqCDDwAAAW+2QhKqAAAEAwBHMEUCIBDMYim2\n\
sF8eHpW1Z7/yQ1liTwa8IRSjidBd9ZVIwe6mAiEA7DPOTaRgc/cH3OzIGSu6dLae\n\
e5F/YRkmC9TikWiWTC8wDQYJKoZIhvcNAQELBQADggEBAIIERQYarXyjBoi2yjhr\n\
7WTjbsPLTUSUFVK2f+qckuDJfoX9bW1PLShve5R8WWgIBu4eZYyUDS+BzXXSV2wg\n\
NscAL9TkxobI/N0HiI5iGtJuI8dVIsFSRMG9IWRz96/pqRcTMz5GlIhAurB3aR+S\n\
MnwAYsrNBHG+rqgUwNVn0h2XoVJe3VxrV2QwTH5kzBwG/Ju1+Khqkvs+9/M3UrJg\n\
qPGwissH0W8HgYWhKVISkN2ui55RgbHXQHYDX0uYgGK6iRMxHHlOR1vOqRgEvVGF\n\
5g8CcK3EzritKaHkD6bf0pnSE/E7cjKXzgB4l+58dsNcIVo9YgID4xYGS6paetso\n\
XHE=\n\
-----END CERTIFICATE-----

johnanderson81, 2022-03-03:
Any resolution to this?  I'm having the same issue with HTTPS on the SIM7070G -- can't ever get the connection open.

rhr407, 2022-04-18:
I receive no reponse for HTTP get request for `http://httpbin.org/get` URL. Following is the output using the AT Command Tester from `https://m2msupport.net`

`Checking registration status...

AT+CREG?

+CREG: 2,1,"912","3D73",0

OK
The device is registered in home network.

AT+CGREG?

+CGREG: 2,1,"912","3D73",0,"1"

OK
The device is registered in home network.

Device is registered.. 

Check the network APN...

AT+CGNAPN

+CGNAPN: 0,""

OK
Network did not send APN to the device.
Activate the network bearer...

AT+CNACT=0,1

OK

+APP PDP: 0,ACTIVE
Set up the HTTP URL...

AT+SHCONF="URL","httpbin.org"

OK
Set up the HTTP body length...

AT+SHCONF="BODYLEN",1024

OK
Set up the HTTP header length...

AT+SHCONF="HEADERLEN",350

OK
Initiating HTTP connection...

AT+SHCONN

OK
Get the HTTP connection state...

AT+SHSTATE?

+SHSTATE: 1

OK
HTTP connection is successful...

HTTP get request...

AT+SHREQ="http://httpbin.org/get",1

OK
No reponse received..

`

baconcheese113, 2022-11-29:
I'm also having problems using HTTPS with my SIM7000G, HTTP works fine using both the AT+HTTPACTION and the AT+SHCONN command suites, but HTTPS always fails at the connection step.
```
AT
OK
AT+CFSWFILE=3,"httpbin_root_ca.cer",0,1077,10000
DOWNLOAD

OK
AT+CFSTERM
OK
AT+CSSLCFG="convert",2,"httpbin_root_ca.cer"
OK
at+csslcfg="sslversion",1,3
OK
AT+SHSSL=1,"httpbin_root_ca.cer"
OK
AT+SHCONF="URL","https://httpbin.org"
OK
AT+SHCONF="BODYLEN",1024
OK
AT+SHCONF="HEADERLEN",350
OK
AT+SHCONN
ERROR
AT+SHSSL=1,""
OK
AT+SHCONN
ERROR
```

One thing you might want to try is removing all the line breaks `\n`s from your certificate. You can validate that it was uploaded successfully by checking QPST EFS Explorer with a direct USB connection to the modem
[image]