botletics/SIM7000-LTE-Shield #202: Example walkthrough for a HTTPS POST and GET

krish2487, 2020-10-06:
Hello,

I know it is a rather vague question, but I was wondering if I can trouble the kind folks here 
for an example walkthrough of a HTTPS POST example for the SIM7000 shield.

I am able to do a HTTP POST and GET request, but looking at the issues on github page,
it looks like the information to do a HTTPS post is scattered and incomplete.
I did go through the docs folder and went through the SSL app note. However, there is 
no single end-to-end example for performing a HTTPS transaction.

I m sure, this information would be helpful for many others like me who are scrambling
in the dark about this specific requirement.

What I am looking for / hoping to get
1. A simple example showing HTTPS POST
2. showing HTTPS GET
3. The pre-requisites ( Firmware on the SIM7000 module, etc etc)
4. Tools ( Software to update the firmware, etc etc)
5. Certificates ?? ( Relevance of certificates and how to obtain them, upload them, software to upload them to the SIM module, how to use them with the AT commands set)

I apologize if it looks like its too much work, but I feel this is pretty helpful for many others like me.
I see a lot of posts with specific sections of information, but no single example that others can use and emulate and learn from..

Thanks in advance! :-)

krish2487, 2020-10-19:
bump!! Any suggestions or inputs?

andrewhong5297, 2020-11-01:
bump looking for similar thing!

botletics, 2021-01-07:
Hi there, I just added some revisions from a pull request adding SSL to the library. Please check out the [fona.addRootCA()](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Library-Functions#fonaaddrootca) function.

richcj10, 2022-04-10:
> I don't think we need to download the certs. Let's assume that I am posting to my own rest api. Let's assume that I trust the certs. Is there a simple modification to the following starting block of code from `Botletics_SIMCom_Library_v1.1.0/Adafruit_FONA.cpp` that would make it possible to post data?
> 
> ```
> 1916 /********************************* HTTPS FUNCTION *********************************/
> 1917 boolean Adafruit_FONA::postData(const char *server, uint16_t port, const char *connType, const char *URL, const char *body) {
> ```

Did you get HTTPS working? I get an error at AT+CHTTPSSTART.

johnedstone, 2022-04-11:
I finally gave up on this board - you can see my trials at [this link](https://github.com/botletics/SIM7000-LTE-Shield/issues/280)  I just felt like the depth of support was not there :(

I do HTTPS POST with JSON for 3G now with [the Arduino MKR GSM 1400](https://github.com/johnedstone/mkrgsm1400-post-json-ssl) and I do 4G LTE with [the GPy pycom board](https://github.com/johnedstone/pycom-gpy)  The Arduino MKR NB 1500 was a bust for me for 4G LTE

richcj10, 2022-04-11:
I actually  got it working last night. The GPy was a bust for me. To many reliability issues withe LTE modem hanging. (Older FW worked way better than newer FW)

johnedstone, 2022-04-11:
Interesting to know, thanks!

richcj10, 2022-04-11:
Take a look at issue https://github.com/botletics/SIM7000-LTE-Shield/issues/292

johnedstone, 2022-04-11:
Wow, nice!  The GPy went so smooth for me that I never looked back.  And eventually, I combined a [GPy  connected to a Pytracker and a Adafruit BME680 reporting ](https://github.com/johnedstone/pycom-gpy/blob/main/projects/post_json_data/main.py) out to a rest api  - so I didn't look back

botletics, 2022-09-08:
Hey guys, figured out how to connect with SSL without verifying certs on the SIM7000! Please do the following:
- Set "SSL_FONA" to 1 in the .h file (remember to save the file before closing)
- Download the latest .cpp file from GitHub
- Open the unedited LTE_Demo example sketch and change "http://dweet.io" to "https://dweet.io" on line 1035 (under the '2' option)
- Upload to the Arduino
- Run the 'G' command to enable data connection, then run '2' to send data to dweet.io using SSL. And voila! 😊