botletics/SIM7000-LTE-Shield #207: MQTT with Sim 7000C NB-IoT Network

najabhaider96, 2020-12-08:
I'm trying to use MQTT on Sim 7000C with NB-IoT Network
I can successfully configured my network with this command

AT+CNACT=1,"apn"

and get IP address with this command

AT+CNACT? 

Then I'm trying to set parameters of MQTT like broker, port, user, password with these command 

AT+SMCONF="URL","m21.cloudmqtt.com:19279"
AT+SMCONF="USERNAME", "user"
AT+SMCONF="PASSWORD", "pass" 

But when I trying to establish MQTT connection with my broker with this command

AT+SMCONN

Module stuck and gives nothing.

What I'm doing wrong please any suggestion for me doing this?

Module firmware version: 1351B05SIM7000C

Rick-GO, 2021-03-10:
Hi @najabhaider96 

I faced the same issues with the 7000E. When I used an IP address instead of a DNS address (from the broker), it worked for me. When I restarted the connection a second time, the DNS was working (strange behavior).

najabhaider96, 2021-04-28:
Hi @Rick-GO 

When I used this https://www.emqx.io/mqtt/public-mqtt5-broker public MQTT Broker with these set of commands without any Username and Password it worked for me.
But I can't use this in my production I don't have any option to use any broker without Username and Password authentication in my production. :(

Here is my commands with responses.
AT+SMSTATE?
+SMSTATE: 0
OK

AT+CSQ
+CSQ: 22,99

OK

AT+CGDCONT=1,"IP","APN" 
OK

AT+CNACT=1,"APN"
OK

+APP PDP: ACTIVE

AT+SMCONF="URL","broker.emqx.io:1883" 
OK

AT+SMCONF="CLIENTID","Device001" 
OK

AT+SMCONF?
+SMCONF 

CLIENTID: "Device001"
URL: broker.emqx.io:1883
KEEPTIME: 60
USERNAME: ""
PASSWORD: ""
CLEANSS: 0
QOS: 0
TOPIC: ""
MESSAGE: ""
RETAIN: 0

OK

AT+SMCONN
OK

AT+SMSUB="update",1
OK

AT+SMPUB="update","5",1,1
> hello

OK

+SMSUB: "update","hello"

AT+SMPUB="update","8",1,1
> HI Najaf

OK

+SMSUB: "update","HI Najaf"