botletics/SIM7000-LTE-Shield #209: sleep while connected to MQTT server

jaka87, 2021-01-06:
Once i establish connection to MQTT server module doesn't go to sleep anymore. I'm not sure if this is normal/expected behavior or not.
Has anybody been able to put module to sleep while connected to MQTT server ?

botletics, 2021-01-31:
This is totally normal, because in sleep mode there is no data connectivity. You can read more about eDRX and sleep modes in the [low power mode application note](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/Technical%20Documents)

closed 2021-01-31 by botletics

jaka87, 2021-01-31:
i dont think what you are saying is 100% true but i think i get what you mean. After extensive testing i believe that MQTT is not the best use for devices like mine (weather station). COAP on the other hand is much better suited for it but its not that well documented. At the end i ended up designing my own protocol based on UDP and had excellent result with it.I decreased payload just to few bytes, and module is connected all the time and can be put to "normal" sleep.

PS: I also manage to get PSM to work. I could easily put it into PSM but had troubles to wake module up. It turns out that i have to restart serial connection and then it worked... Again it is not well suited for my use since i send data every 5-10min and waking up module takes about 10 extra seconds...

thesoaringpotato, 2021-03-21:
@jaka87 Would it be possible to briefly post how you got the module to enter PSM?  I can confirm with AT+CEREG? and AT+CPSMS? that the TAU and active timers are set.  Still, the module refuses to sleep.

AT+CEREG?
RESPONSE -> +CEREG: 4,5,,,,,,,"00001111","01011111"
RESPONSE -> OK
AT+CPSMS?
RESPONSE -> +CPSMS: 1,,,"01011111","00001111"
RESPONSE -> OK

Restart the module with AT+CFUN=6, module restarts.

Module boots, registers on network, and sits idle.  But never enters PSM.

Thank you!

jaka87, 2021-03-22:
I dont remember 100% becouse i stoped using it...

I ttink I had to set set sleep
`fona.enableSleepMode(true);`
then 
`fona.enablePSM(true, "00100001", "00100011");`

I also changed DTR pin state to enter sleep. Not sure if this was mandatory...

thesoaringpotato, 2021-03-22:
Hey @jaka87 thanks for your quick reply, I certainly appreciate it!  Turns out the modem is able to fall asleep.  The timing on which the modem falls to idle, and then falls asleep is a bit fluid, ranging from 1-15 minutes.  I'll need to do more work to see what is keeping the modem awake.  Nonetheless I am now able to place the modem in PSM mode on a consistent basis.

Cheers!  Also that weather station looks great.

jaka87, 2021-03-24:
Im glad you sorted it out. If modem doesn't go to sleep it may be
because it is pinging the server. If you are using client to server
connection i would suggest you dont use MQTT but rather use something like
COAP that works on UDP instead of TCP. I made a custom protocol similar to
COAP but simpler. You can check my git, i have published client and also
server code in python...

Thanks. I been working hard on implementation...

coratron, 2021-11-15:
@thesoaringpotato Would you be able to share your flow to enter PSM? We'd really appreciate it.

thesoaringpotato, 2021-11-15:
My apologies, in the end I had to abandon the project since I couldn't place the modem to sleep consistently.  The modem now is turned off and on depending on frequency of updates with my stations.

However, and this may aid your troubleshooting, is that the ability for the modem to sleep may in fact be dependent on the carrier.  I'm solely using Hologram and didn't have an alternate carrier to test this theory unfortunately.