botletics/SIM7000-LTE-Shield #211: compatibility with LILYGO® TTGO T-SIM7000G

Macfly, 2021-01-14:
anyone has try to make this code work with the LILYGO® TTGO T-SIM7000G Module ESP32-WROVER-B board?

https://www.aliexpress.com/item/1005001625046691.html?spm=a2g0o.cart.0.0.456e3c00qx7Cgj&mp=1

DrMarty, 2021-01-27:
Hi Macfly,
I've started getting this working on TTGO-T-SIM7000G: https://github.com/DrMarty/TTGO-T-SIM7000x
At this point, it's still a bit of a hack as I haven't linked to the Botletics repo (i've just copied the code directly into the project).
I'll be trying to clean this up over the next few days and fleshing it out to include more functionality.

Macfly, 2021-01-27:
thanks for the response.
I'll check it out!

DrMarty, 2021-02-01:
A quick update...
I've just pushed my lastest updates to my github repository (see above).
It's still very bare-bones but I'm much happier with the structure. Specifically I've removed all references to Botletics and Adafruit_FONA. There were a number of things about the FONA code that I didn't like and because Botletics is built on top of that, changing one meant ripping gaping holes in the other. In the end the FONA code has far too much overlap between the SIMCOM hardware and the AT command processing, so I've abstracted them apart from each other and created two separate classes - SIM7000x and ATCOM. I plan to create my own PCB and possibly use alternate modem chips so I wanted to keep the hardware and AT command processors separated.

Macfly, 2021-02-01:
great!
what modem chip are you thinking about? SIM7070G?

DrMarty, 2021-02-01:
Actually i haven't put a whole lot of thought into it yet. I picked the SIM7000G because it has global band coverage, LTE Cat-M1, NB-IoT as well as GPS  / GNSS and this gives me the broadest possible combination of features I may use. But if I can drop one or a couple of those features and go down in price then maybe I will.  I also want the option of going with Quectel or U-Blox

Macfly, 2021-03-06:
I have time finally but your repo looks down. Did you give up trying to make it work?

DrMarty, 2021-03-08:
@Macfly Am still working on it but I've had to 'close source' it for now until we get our product out the door.

ishdemon, 2021-08-28:
@DrMarty ah bummer. I am using same board but having trouble getting it to work.. I am unable to get the board detected by the library even though I have set the correct RX TX pins.. It always fails saying `could not found Fona` in modulesetup