botletics/SIM7000-LTE-Shield #213: Adafruit stops receiving data after sometime.

Alfredfoby, 2021-01-22:
So I am having an issue where I am publishing values to adafruit lets say every 30 seconds. The issue I'm facing is that it would send data to adafruit io for maybe a few hours, sometimes a few days then stop sending data completely and randomly send one piece of data to adafruit io and stop again for days and days.

I put a bunch of print statements and realized it is stuck in the while loop doing the mqtt.connect() function and I am not sure what to do to fix it. I initially thought it was an issue with the modem and hologram sim card data plan but that doesn't seem to be the case as I talked to hologram already.

This is the loop I am talking about. 

`while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
      Serial.println(mqtt.connectErrorString(ret));
      Serial.println("Retrying MQTT connection in 5 seconds...");
      mqtt.disconnect();
      delay(5000);  // wait 5 seconds
    }
    Serial.println("MQTT Connected!");`

It would be of great help as I am blind to monitoring my devices.

botletics, 2021-01-22:
Worst case, you could put a counter variable in the while loop that is incremented every time the reconnection fails, then when counter reaches XX number of attempts, you restart the SIM7000 module using AT+CFUN=6 to do a software reset.

Alfredfoby, 2021-01-22:
> Worst case, you could put a counter variable in the while loop that is incremented every time the reconnection fails, then when counter reaches XX number of attempts, you restart the SIM7000 module using AT+CFUN=6 to do a software reset. Something like this:
> 
> ```
> uint8_t counter = 0;
> 
> while ((ret = mqtt.connect()) != 0) {
>   Serial.println(mqtt.connectErrorString(ret));
>   Serial.println("Retrying MQTT connection in 5 seconds...");
>   counter++; // Increment counter
>   mqtt.disconnect();
>   if (counter == 10) {
>     fona.setFunctionality(6); // AT+CFUN=6 to soft restart the SIM7000
>   }
>   else {
>     delay(5000);
>   }
> }
> 
> Serial.println("MQTT Connected!");
> ```

I have done that but theres still no luck with publishing to adafruit. I have ran out of ideas.

botletics, 2021-01-22:
Is it losing network connection altogether?

Alfredfoby, 2021-01-22:
> Is it losing network connection altogether?

Yes I believe so,  in mqtt.connect() function it never seems to connect. Could it have something to do with mqtt.keepalive() window?

botletics, 2021-01-22:
- Do you have a LiPo battery connected? (Make sure red wire is on the left side when you plug it into the connector)
- So even after a soft restart, the module doesn't connect to the network? What do you typically do to get it connected again? - You should probably put a network registration (with getNetworkStatus) before even attempting to connect via MQTT. What signal strength do you typically get when it's connected?
- What if you did something like this instead of soft-restarting the SIM7000:

```
uint8_t counter = 0;

while ((ret = mqtt.connect()) != 0) {
  Serial.println(mqtt.connectErrorString(ret));
  Serial.println("Retrying MQTT connection in 5 seconds...");
  counter++; // Increment counter
  mqtt.disconnect();
  if (counter == 10) {
    // fona.setFunctionality(6); // AT+CFUN=6 to soft restart the SIM7000

    fona.setFunctionality(1); // AT+CFUN=1
    fona.setNetworkSettings(F("hologram"));
    fona.enableGPRS(true); // Turn on data
    delay(2000); // Pause before retrying to connect
  }
  else {
    delay(5000);
  }
}

Serial.println("MQTT Connected!");
```

Alfredfoby, 2021-01-22:
> * Do you have a LiPo battery connected? (Make sure red wire is on the left side when you plug it into the connector)
> * So even after a soft restart, the module doesn't connect to the network? What [...]

So i am doing all that you have in your code already with the exception of fona.enableGPRS() because I am not using the gps side of things. 

- I have the LiPo battery connected already
- I don't do anything to connect it again, it typically connects on its own then works for a while and stops again. or if i leave the system off for days then it works for a bit and stops.
- I am not really sure the signal strength but I used the signal quality at command,  AT+CSQ and i got 20,99 which is excellent i believe.
-

botletics, 2021-01-22:
enableGPRS is for data connection, not GPS.

Alfredfoby, 2021-01-22:
> enableGPRS is for data connection, not GPS. Also, you could try sending "AT&F0" one time to factory reset the module and run your usual code again.

I will try that soon and give you an update, I am inclined to believe it has something to do with the hologram network not connecting to the internet. But I may be wrong as well.

Alfredfoby, 2021-01-26:
> enableGPRS is for data connection, not GPS. Also, you could try sending "AT&F0" one time to factory reset the module and run your usual code again.

Okay i think I need to send enableGPRS before it actually sends the data.
It is working currently and I will continue to update you, Thanks for the help.

closed 2021-01-31 by botletics

Alfredfoby, 2021-02-01:
> Okay i think I need to send enableGPRS before it actually sends the data.
> It is working currently and I will continue to update you, Thanks for the help.

So it is currently not working anymore, it is almost as if a buffer or something gets full on the modem and it needs some sort of reset because it completely stopped on its own and I've been trying to get it to send data even after resetting the modem for the past 3 days

Alfredfoby, 2021-02-02:
> Okay i think I need to send enableGPRS before it actually sends the data.
> It is working currently and I will continue to update you, Thanks for the help.

I figured that every time it stops communicating I would do a hard reset of the modem. And that seems to be working when i send AT&F0 and reset the modem.

Is there a way to do this with arduino code maybe using the reset pin??

botletics, 2021-02-02:
Yes, you can use the `FONA_RST` pin in the Arduino code to reset it by pulsing it LOW for at least 252ms, according to Table 10 of the SIM7000 hardware design doc. You can also invoke the AT+CFUN=6 command by using the Arduino code `fona.setFunctionality(6)`