botletics/SIM7000-LTE-Shield #217: Unable to build LTE_Demo example, HTTP_GET

aimeeraymond, 2021-01-30:
Have downloaded latest version of Adafruit_FONA library.  Am getting an error when building the LTE_Demo script

Compiling library "Adafruit_FONA"
"C:\\Users\\aimee\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\aimee\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino/api/deprecated" "-IC:\\Users\\aimee\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino" "-IC:\\Users\\aimee\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\variants\\uno2018" "-IC:\\Users\\aimee\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\libraries\\SPI\\src" "-IC:\\Users\\aimee\\Documents\\Arduino\\libraries\\WiFiNINA\\src" "-IC:\\Users\\aimee\\Documents\\Arduino\\libraries\\Adafruit_FONA" "-IC:\\Users\\aimee\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\libraries\\SoftwareSerial\\src" "C:\\Users\\aimee\\Documents\\Arduino\\libraries\\Adafruit_FONA\\Adafruit_FONA.cpp" -o "C:\\Users\\aimee\\AppData\\Local\\Temp\\arduino_build_717024\\libraries\\Adafruit_FONA\\Adafruit_FONA.cpp.o"
C:\Users\aimee\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA_LTE::HTTP_connect(const char*)':
C:\Users\aimee\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:2074:45: warning: NULL used in arithmetic [-Wpointer-arith]
   if (strcmp(replybuffer, "+SHSTATE: 1") == NULL) return false;
 
The code section the error is point to is:

// Get HTTP status
  getReply(F("AT+SHSTATE?"));
  if (strcmp(replybuffer, "+SHSTATE: 1") == NULL) return false;
  readline(); // Eat 'OK'

A response of 0 means disconnected, a response of 1 means connected.
The HTTP_GET and HTTP_POST functions may be related to the issue.  They are changed from the prior version with the use of  a variable "URI" versus "URL"

botletics, 2021-01-30:
Please see the latest version of the library in the master branch. The library and LTE_Demo example was updated today.

aimeeraymond, 2021-01-31:
Thanks.  I have updated the library.  There is still a warning message, but the script compiles and runs as expected.

closed 2021-01-31 by aimeeraymond