botletics/SIM7000-LTE-Shield #218: Question:  How to configure to forward LoRaWan Gateway packets to TTN

aimeeraymond, 2021-01-31:
I am working on a project to get sensor data to the TTN server.  The process flow is:
A sensor node (Heltec Cubecell Capsule) collects the data value and sends it to a LoRa gateway (Heltec HT-M00).  
The LoRa gateway connects via WiFi using UDP (server = router.us.thethings.network, port=1700) to send each packet to the TTN server.  The TTN server sends an acknowledgement back to gateway.

This process flow works if I use my home WiFi (running on a MoFI4500 modem), but that is not real-life.
The sensors are actually placed in agriculture fields with no WiFi or cellular networks around, hence the use of LoRaWan.  I need to replace my home WiFi modem with the Arduino/SIM7000.  The Arduino/SIM7000 will be placed outdoors within WiFi range of the LoRa gateway.

In my Arduino script, in the set-up 
  1. get the Cellular connection
         fonaSS.begin(9600)
         fona.getIMEI
        fona.setFunctionality(1)
        fona.setNetworkSettings(F("hologram"))
        fona.setPreferredLTEMode(1)  // use LTE CAT-M
        fona.enableRTC(true)
        fona.enableGPS(true)
       fona.enableGPRS(true)
       fona.getGPS(lat, long, ...)
  2. create a WiFi Access Point
      WiFi.config(IPAddress(192.168.44.1))
      WiFi.beginAP(ssid, password)
  
In the loop
  If status = WL_AP_CONNECTED, 
        then print "Device connected to AP", 
   else "Device disconnected"
end

When I run the script, the LoRa gateway will connect to the WiFi Access Point (FlumelogicAP) and reports "Host esp32-xxxx WiFi connected to ssid on IP=192.168.44.2 Local UDP port=1700, Connection successful

Heltec HT-M00 firmware Rev 1.4

WiFi SSID   :  FlumelogicAP
WiFi PASS   :  xxxxxxxx
CH0 FREQ    :  903900000
CH1 FREQ    :  904100000
MIN SF      :  7
MAX SF      :  12
SERVER ADDR :  13.66.213.36
SERVER PORT :  1700
TIME ZONE   :  UTC-8
GateWayID   :  7C9EBDFFFFFC2A30

Connecting to FlumelogicAP
get time from : europe.pool.ntp.org
get time from : asia.pool.ntp.org
get time from : oceania.pool.ntp.org
get time from : north-america.pool.ntp.org
get time from : south-america.pool.ntp.org
get time from : africa.pool.ntp.org

Host esp32-fc2a30 WiFi Connected to FlumelogicAP on IP=192.168.44.2
Local UDP port=1700, Connection successful
stat update: <113> {"stat":{"time":"1970-01-01 00:00:20 CET","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}

stat update: <113> {"stat":{"time":"1970-01-01 00:01:00 CET","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}

Json Up: {"rxpk":[{"tmst":62388735,"chan":1,"rfch":1,"freq":904.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":16,"rssi":-17,"size":23,"data":"QMolAiYADwIC5zpEpdXLvpUDhg0IGQI="}]}

But the LoRa gateway can't seem to resolve the ntp time server or the TTN server name, which seem to be an inability to handle the DNS.  The LoRa gateway is getting Json packets.  I don't have access to the LoRa gateway scripting

When connected to the house WiFi (Flumelogic), the packets are getting forwarded without issue,

Heltec HT-M00 firmware Rev 1.4

WiFi SSID   :  Flumelogic
WiFi PASS   :  yyyyy
CH0 FREQ    :  904500000
CH1 FREQ    :  904700000
MIN SF      :  7
MAX SF      :  12
SERVER ADDR :  router.us.thethings.network
SERVER PORT :  1700
TIME ZONE   :  UTC-8
GateWayID   :  7C9EBDFFFFFC29B8

Connecting to Flumelogic
get time from : europe.pool.ntp.org
Time synchronized : 2021-01-31 22:06:52 CET

Host esp32-fc29b8 WiFi Connected to Flumelogic on IP=192.168.10.77
Local UDP port=1700, Connection successful
stat update: <113> {"stat":{"time":"2021-01-31 22:06:52 CET","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}

Json Up: {"rxpk":[{"tmst":10685310,"chan":0,"rfch":0,"freq":904.500000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":15,"rssi":-48,"size":23,"data":"QMolAiYAKgICVgESmcC9cbypP5BWFow="}]}
Json Up: {"rxpk":[{"tmst":27103392,"chan":1,"rfch":1,"freq":904.700000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":17,"rssi":-17,"size":23,"data":"QMolAiYAKwICCYR7HG72yyZ0Ss3k8Us="}]}

>> What am I missing here?  How do I forward the packets?  From the perspective of the SIM7000, it doesn't know who sent the packet or where the packet is going, so it is not a HTTP thing.  I am thinking it is a TCP/UDP thing, but not clear how to transact the packets.

botletics, 2021-01-31:
Have you taken a look at the [TCP functions](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Library-Functions#fonatcpconnect)?

aimeeraymond, 2021-02-02:
Using TCP appears to be the correct option.  I didn't realize that Hologram is actually handling the data package and routing it.  One is actually sending a message to the Hologram Cloud, https://www.hologram.io/references/embedded-apis.

closed 2021-02-02 by aimeeraymond