botletics/SIM7000-LTE-Shield #219: Http GET/ Libraries

wilba84, 2021-02-01:
Hey,
I have been using a earlier version of this library with a sim7000e and HTTP GET, With no issues since.
I have upgraded to a sim7600ce-t and having a few issues. I dont seem to have any issues with the module or registering on the network, The errors arrive when posting to the api, likely something has been missed while upgrading the library.

These are the errors using sim7600, regardless if its the old or new library or if I #Define SIM7000 / 7500 / 7600

[image]
[image]
[image]

Below this is a successful post using sim7000e and the old libary

[image]
[image]
[image]

Any ideas? im sure its something small
Cheers
Will

botletics, 2021-02-01:
Hi there,

First of all, SIM7600 does not support NB-IoT, so your config for that shouldn't be there. But I think the problem is that you may be using option '2' instead of option '3' in the LTE_Demo to post data. Option '2' is for 2G and SIM7000/7070, whereas '3' is for SIM5320/7500/7600 because they share the same commands.

wilba84, 2021-02-01:
Hi,
I have Removed;
    fona.setPreferredLTEMode(2);
    fona.setOperatingBand("NB-IOT", 28);

Then I had a look through the library and tried
    fona.setPreferredLTEMode(1);
    fona.setOperatingBand("CAT-M", 28);

Which also returns errors. So i just removed the commands entirely and it connects to the network fine.

Im actually using the GET part of the IOT example sketch but I will use the SIM7XXX_LTE_DEMO until we work out what is wrong. 

I can also confirm that I have
#define SIMCOM_7600
//#define SIMCOM_7000
And that the SIM7600 is detected.

I followed the sim7500 shield demo you linked, and this was the result.
[image]

This is the section of code relating to the "3" command. I simplified it somewhat just for troubleshooting
[image]

I started looking at fona,postdata command in the library. 
[image]

I couldnt find "AT+CHTTPSOPSE" or "AT+CHTTPSSTART" in the SIM7500/7600 AT Commands Set, however its in the library
https://simcom.ee/documents/SIM7600C/SIM7500_SIM7600%20Series_AT%20Command%20Manual_V1.01.pdf

I think this is whats making it fail, any ideas?

botletics, 2021-02-02:
You need to use "G" before "3"

wilba84, 2021-02-02:
[image]

wilba84, 2021-02-02:
Not sure if its related but using the new library, sim7000, everything set to sim7000, ANY demo sketch from the library makes sim7000 unable to register on the network. If i revert to the old one it is fine

botletics, 2021-02-02:
What if you comment out line 1967 from Adafruit_FONA.cpp? It's this line: `if (strcmp(replybuffer, "+CHTTPSOPSE: 0") != 0) return false;`

wilba84, 2021-02-02:
I commented this out
[image]
This is the serial out, It says it failed, but it worked
[image]

wilba84, 2021-02-02:
With that change both LTE_Example and IOT_Example post to dweet fine using sim7600 and new library.
I need to change the post format to suit sim7600. Below you can see the new code defined for sim7600 and the code for the old format commented out below
[image]
This was the serial out
[image]
It doesnt post successfully, did I fill the format code out correctly?

botletics, 2021-02-02:
You should use port 443 for HTTPS, not port 80.

botletics, 2021-02-02:
I just updated the Adafruit_FONA.cpp file on GitHub. Can you use that file and see if it runs fine now?

wilba84, 2021-02-02:
This is the 1.0.3 library with the .cpp that you uploaded for me to try. I also changed "HTTPS" to "HTTP"
[image]

botletics, 2021-02-02:
Use 80 for HTTP and 443 for HTTPS. Please try the new Adafruit_FONA.cpp file I revised, because it looks like the commands are running just fine but the code thinks it's not.

wilba84, 2021-02-02:
that was with the new Adafruit_FONA.cpp

wilba84, 2021-02-02:
If I comment out 
"if (strstr(replybuffer, "+CHTTPSSEND: 0") == NULL) return false;"
"if (strstr(replybuffer, "+CHTTPSOPSE: 0") == NULL) return false;"
(The ones im guessing you put in)
And allow the serial to run, I get this response
[image]

botletics, 2021-02-03:
Sorry, I meant to try it on dweet.io to make sure it works first before changing it to another server.

wilba84, 2021-02-03:
This is dweet.io
[image]

wilba84, 2021-02-03:
This is very close
[image]
Does the circled URL refer to "sprintf(URL" that is above it? It seems to be a format issue with the post. But Im not sure what the actual post ends up looking like when its strung together
[image]

botletics, 2021-02-03:
There are two places where you need to comment out the strcmp check for "+CHTTPSSEND: 0" if you make that mod.
And yes, the URL is constructed using the sprintf() above it.

botletics, 2021-02-03:
Also, you might want to change the getReply timeout to 10000 instead of 2000:

```
// Get server response content
sprintf(auxStr, "AT+CHTTPSRECV=%i", replyLen);
getReply(auxStr, 10000);
```

wilba84, 2021-02-03:
Ok, I have made those changes.
So basically its saying Connect to api.weathercloud.net on Port 80 using HTTP and send "http://api.weathercloud.net/set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd/temp/%s"
Or is it  
(GET?)(http://) (api.weathercloud.net) (:80) (http://api.weathercloud.net/set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd/temp/%s)
As in the 4/5 elements All strung together?

wilba84, 2021-02-03:
I was soo close with the above post, I was missing a slash and formatted slightly incorrrect.
Very happy, its posting and making it to the api
[image]

wilba84, 2021-02-03:
For me to get it working I had to comment out both pairs of these(4)
//if (strcmp(replybuffer, "+CHTTPSSEND: 0") != 0) return false;
//if (strstr(replybuffer, "+CHTTPSSEND: 0") == NULL) return false;

This did not make a difference - changed back to 2000
sprintf(auxStr, "AT+CHTTPSRECV=%i", replyLen);
getReply(auxStr, 10000);

This is the Serial out of a successful post, Still several commands showing errors, and it says it failed all 3 times.

> 15:09:05.464 -> 
15:09:05.464 -> 
15:09:05.511 -> 	---> AT+CGREG?
15:09:05.511 -> 	<--- +CGREG: 0,1
15:09:05.511 -> Network status 1: Registered (home)
15:09:05.511 -> Network services Connected!
15:09:08.571 -> 	---> AT+CHTTPSSTART
15:09:08.571 -> 	<--- OK
15:09:08.571 -> Waiting 1s to ensure connection...
15:09:09.594 -> 	---> AT+CHTTPSOPSE="api.weathercloud.net",80,1
15:09:09.594 -> 	<--- OK
15:09:10.301 -> 	<--- +CHTTPSOPSE:0
15:09:10.301 -> Waiting 1s to make sure it works...
15:09:11.335 -> 	---> AT+CHTTPSSEND=268
15:09:21.454 -> 	<--- >
15:09:21.501 -> 	---> GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd/temp/0.0/hum/0/bar/0/rain/0.0/tempin/0/humin/0/dewin/0/heatin/0/rainrate/0.0/wdir/158/wspd/0/wspdavg/0/wspdhi/0/wdiravg/1/dew/NAN/uvi/6553.5/chill/21/heat/-178/ HTTP/1.1
15:09:21.501 -> Host: api.weathercloud.net
15:09:21.501 -> 
15:09:21.501 -> 
15:09:21.547 -> 	<--- OK
15:09:21.690 -> 	<--- +CHTTPSSEND:0
15:09:22.724 -> 	---> AT+CHTTPSRECV?
15:09:22.724 -> 	<--- +CHTTPSRECV: LEN,0
15:09:22.771 -> 	---> AT+CHTTPSRECV=0
15:09:22.771 -> 	<--- ERROR
15:09:22.818 -> 	---> AT+CHTTPSCLSE
15:09:22.818 -> 	<--- +CHTTPSCLSE:0
15:09:22.818 -> Failed to complete HTTP/HTTPS request...
15:09:23.854 -> 	---> AT+CHTTPSSTART
15:09:23.854 -> 	<--- ERROR
15:09:23.854 -> Waiting 1s to ensure connection...
15:09:24.910 -> 	---> AT+CHTTPSOPSE="api.weathercloud.net",80,1
15:09:24.910 -> 	<--- OK
15:09:25.379 -> 	<--- +CHTTPSOPSE:0
15:09:25.379 -> Waiting 1s to make sure it works...
15:09:26.411 -> 	---> AT+CHTTPSSEND=268
15:09:36.523 -> 	<--- >
15:09:36.570 -> 	---> GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd/temp/0.0/hum/0/bar/0/rain/0.0/tempin/0/humin/0/dewin/0/heatin/0/rainrate/0.0/wdir/158/wspd/0/wspdavg/0/wspdhi/0/wdiravg/1/dew/NAN/uvi/6553.5/chill/21/heat/-178/ HTTP/1.1
15:09:36.570 -> Host: api.weathercloud.net
15:09:36.570 -> 
15:09:36.570 -> 
15:09:36.617 -> 	<--- OK
15:09:36.757 -> 	<--- +CHTTPSSEND:0
15:09:37.789 -> 	---> AT+CHTTPSRECV?
15:09:37.789 -> 	<--- +CHTTPSRECV: LEN,202
15:09:37.836 -> 	---> AT+CHTTPSRECV=202
15:09:37.836 -> 	<--- OK
15:09:37.930 -> 	<--- 
15:09:37.930 -> +CHTTPSRECV: DATA,202
15:09:37.930 -> http/1.1 200 ok
15:09:37.930 -> server: nginx
15:09:37.930 -> date: wed, 03 feb 2021 04:09:37 gmt
15:09:37.930 -> content-type: text/html; charset=utf-8
15:09:37.930 -> transfer-encoding: chunked
15:09:37.930 -> connection: keep-alive
15:09:37.930 -> x-powered-by: ph
15:09:37.978 -> 	---> AT+CHTTPSCLSE
15:09:37.978 -> 	<--- +CHTTPSCLSE:0
15:09:37.978 -> Failed to complete HTTP/HTTPS request...
15:09:39.053 -> 	---> AT+CHTTPSSTART
15:09:39.053 -> 	<--- ERROR
15:09:39.053 -> Waiting 1s to ensure connection...
15:09:40.063 -> 	---> AT+CHTTPSOPSE="api.weathercloud.net",80,1
15:09:40.110 -> 	<--- OK
15:09:40.485 -> 	<--- +CHTTPSOPSE:0
15:09:40.485 -> Waiting 1s to make sure it works...
15:09:41.571 -> 	---> AT+CHTTPSSEND=268
15:09:51.640 -> 	<--- >
15:09:51.686 -> 	---> GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd/temp/0.0/hum/0/bar/0/rain/0.0/tempin/0/humin/0/dewin/0/heatin/0/rainrate/0.0/wdir/158/wspd/0/wspdavg/0/wspdhi/0/wdiravg/1/dew/NAN/uvi/6553.5/chill/21/heat/-178/ HTTP/1.1
15:09:51.734 -> Host: api.weathercloud.net
15:09:51.734 -> 
15:09:51.734 -> 
15:09:51.734 -> 	<--- OK
15:09:51.874 -> 	<--- +CHTTPSSEND:0
15:09:52.905 -> 	---> AT+CHTTPSRECV?
15:09:52.952 -> 	<--- +CHTTPSRECV: LEN,202
15:09:52.952 -> 	---> AT+CHTTPSRECV=202
15:09:52.952 -> 	<--- OK
15:09:53.046 -> 	<--- 
15:09:53.046 -> +CHTTPSRECV: DATA,202
15:09:53.046 -> http/1.1 200 ok
15:09:53.092 -> server: nginx
15:09:53.092 -> date: wed, 03 feb 2021 04:09:52 gmt
15:09:53.092 -> content-type: text/html; charset=utf-8
15:09:53.092 -> transfer-encoding: chunked
15:09:53.092 -> connection: keep-alive
15:09:53.092 -> x-powered-by: ph
15:09:53.138 -> 	---> AT+CHTTPSCLSE
15:09:53.138 -> 	<--- +CHTTPSCLSE:0
15:09:53.138 -> Failed to complete HTTP/HTTPS request...
15:09:55.152 -> Waiting for 600 seconds
15:09:55.152 -> 

Also v1.02 and onwards of the library, do not work with sim7000. The module fails to register the sim on the network.

Thank you very much Tim, appreciate your work. Theres plenty of info on this post to help with library updates/bug fixes, let me know if you would like me to try any code out for you, more than happy.

I also want to touch on my other post for the FTP image transfering. I didnt realise it was on your to-do list until after I had posted, however I have ordered the camera and parts, which will arrive soon and I will help with code/tutorial/images where I can

botletics, 2021-02-03:
Your query string is wrong. You should be doing something like "http://api.weathercloud.net/set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd/temp=%s" with sprintf()

wilba84, 2021-02-03:
>     while (counter < 3 && !fona.postData("api.weathercloud.net", 80, "HTTP", URL)) { // Server, port, connection type, URL
Isnt this combining it into that?

botletics, 2021-02-03:
You are formatting URL incorrectly. You are using forward slashes to separate names and values like "temp/12.3/heat/-178", whereas you should be using '&' and '=' like "temp=12.3&heat=-178"

wilba84, 2021-02-03:
Like this?
> `"GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd?temp=%s&hum=%s&bar=%s&rain=%s&tempin=%s&humin=%s&dewin=%s&heatin=%s&rainrate=%s&wdir=%s&wspd=%s&wspdavg=%s&wspdhi=%s&wdiravg=%s&dew=%s&uvi=%s&chill=%s&heat=%s/ HTTP/1.1\r\nHost: api.weathercloud.net\r\n\r\n",

botletics, 2021-02-03:
Yup

wilba84, 2021-02-03:
New format, successful post, same result?

> 15:58:14.511 -> 
15:58:14.558 -> 	---> AT+CGREG?
15:58:14.558 -> 	<--- +CGREG: 0,1
15:58:14.558 -> Network status 1: Registered (home)
15:58:14.558 -> Network services Connected!
15:58:17.608 -> 	---> AT+CHTTPSSTART
15:58:17.608 -> 	<--- ERROR
15:58:17.608 -> Waiting 1s to ensure connection...
15:58:18.667 -> 	---> AT+CHTTPSOPSE="api.weathercloud.net",80,1
15:58:18.667 -> 	<--- OK
15:58:19.172 -> 	<--- +CHTTPSOPSE:0
15:58:19.172 -> Waiting 1s to make sure it works...
15:58:20.229 -> 	---> AT+CHTTPSSEND=268
15:58:30.336 -> 	<--- >
15:58:30.336 -> 	---> GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd?temp=0.0&hum=0&bar=0&rain=0.0&tempin=0&humin=0&dewin=0&heatin=0&rainrate=0.0&wdir=135&wspd=0&wspdavg=0&wspdhi=0&wdiravg=1&dew=NAN&uvi=6553.5&chill=21&heat=-178/ HTTP/1.1
15:58:30.382 -> Host: api.weathercloud.net
15:58:30.382 -> 
15:58:30.382 -> 
15:58:30.382 -> 	<--- OK
15:58:30.523 -> 	<--- +CHTTPSSEND:0
15:58:31.571 -> 	---> AT+CHTTPSRECV?
15:58:31.571 -> 	<--- +CHTTPSRECV: LEN,202
15:58:31.618 -> 	---> AT+CHTTPSRECV=202
15:58:31.618 -> 	<--- OK
15:58:31.712 -> 	<--- 
15:58:31.712 -> +CHTTPSRECV: DATA,202
15:58:31.712 -> http/1.1 200 ok
15:58:31.712 -> server: nginx
15:58:31.712 -> date: wed, 03 feb 2021 04:58:32 gmt
15:58:31.712 -> content-type: text/html; charset=utf-8
15:58:31.748 -> transfer-encoding: chunked
15:58:31.748 -> connection: keep-alive
15:58:31.748 -> x-powered-by: ph
15:58:31.795 -> 	---> AT+CHTTPSCLSE
15:58:31.795 -> 	<--- +CHTTPSCLSE:0
15:58:31.795 -> Failed to complete HTTP/HTTPS request...
15:58:32.815 -> 	---> AT+CHTTPSSTART
15:58:32.815 -> 	<--- ERROR
15:58:32.815 -> Waiting 1s to ensure connection...
15:58:33.852 -> 	---> AT+CHTTPSOPSE="api.weathercloud.net",80,1
15:58:33.899 -> 	<--- OK
15:58:34.320 -> 	<--- +CHTTPSOPSE:0
15:58:34.320 -> Waiting 1s to make sure it works...
15:58:35.340 -> 	---> AT+CHTTPSSEND=268
15:58:45.460 -> 	<--- >
15:58:45.507 -> 	---> GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd?temp=0.0&hum=0&bar=0&rain=0.0&tempin=0&humin=0&dewin=0&heatin=0&rainrate=0.0&wdir=135&wspd=0&wspdavg=0&wspdhi=0&wdiravg=1&dew=NAN&uvi=6553.5&chill=21&heat=-178/ HTTP/1.1
15:58:45.507 -> Host: api.weathercloud.net
15:58:45.507 -> 
15:58:45.507 -> 
15:58:45.554 -> 	<--- OK
15:58:45.696 -> 	<--- +CHTTPSSEND:0
15:58:46.726 -> 	---> AT+CHTTPSRECV?
15:58:46.726 -> 	<--- +CHTTPSRECV: LEN,202
15:58:46.773 -> 	---> AT+CHTTPSRECV=202
15:58:46.773 -> 	<--- OK
15:58:46.819 -> 	<--- 
15:58:46.819 -> +CHTTPSRECV: DATA,202
15:58:46.819 -> http/1.1 200 ok
15:58:46.819 -> server: nginx
15:58:46.819 -> date: wed, 03 feb 2021 04:58:46 gmt
15:58:46.819 -> content-type: text/html; charset=utf-8
15:58:46.865 -> transfer-encoding: chunked
15:58:46.865 -> connection: keep-alive
15:58:46.865 -> x-powered-by: ph
15:58:46.865 -> 	---> AT+CHTTPSCLSE
15:58:46.912 -> 	<--- +CHTTPSRECV:0
15:58:46.912 -> Failed to complete HTTP/HTTPS request...
15:58:47.941 -> 	---> AT+CHTTPSSTART
15:58:47.941 -> 	<--- ERROR
15:58:47.941 -> Waiting 1s to ensure connection...
15:58:48.965 -> 	---> AT+CHTTPSOPSE="api.weathercloud.net",80,1
15:58:48.965 -> 	<--- OK
15:58:49.382 -> 	<--- +CHTTPSOPSE:0
15:58:49.382 -> Waiting 1s to make sure it works...
15:58:50.404 -> 	---> AT+CHTTPSSEND=268
15:59:00.525 -> 	<--- >
15:59:00.571 -> 	---> GET /set/wid/39dd3da8e5a13c9f/key/aef53258d7e2a84f4399a4fadaf641dd?temp=0.0&hum=0&bar=0&rain=0.0&tempin=0&humin=0&dewin=0&heatin=0&rainrate=0.0&wdir=135&wspd=0&wspdavg=0&wspdhi=0&wdiravg=1&dew=NAN&uvi=6553.5&chill=21&heat=-178/ HTTP/1.1
15:59:00.571 -> Host: api.weathercloud.net
15:59:00.571 -> 
15:59:00.571 -> 
15:59:00.618 -> 	<--- OK
15:59:00.759 -> 	<--- +CHTTPSSEND:0
15:59:01.807 -> 	---> AT+CHTTPSRECV?
15:59:01.807 -> 	<--- +CHTTPSRECV: LEN,202
15:59:01.854 -> 	---> AT+CHTTPSRECV=202
15:59:01.854 -> 	<--- OK
15:59:01.900 -> 	<--- 
15:59:01.900 -> +CHTTPSRECV: DATA,202
15:59:01.900 -> http/1.1 200 ok
15:59:01.900 -> server: nginx
15:59:01.900 -> date: wed, 03 feb 2021 04:59:01 gmt
15:59:01.900 -> content-type: text/html; charset=utf-8
15:59:01.946 -> transfer-encoding: chunked
15:59:01.946 -> connection: keep-alive
15:59:01.946 -> x-powered-by: ph
15:59:01.946 -> 	---> AT+CHTTPSCLSE
15:59:01.993 -> 	<--- +CHTTPSCLSE:0
15:59:01.993 -> Failed to complete HTTP/HTTPS request...
15:59:03.986 -> Waiting for 600 seconds
15:59:03.986 ->

botletics, 2021-02-03:
Looks good. Your previous one may have gone through, but it doesn't follow the standard HTTP GET request format.