botletics/SIM7000-LTE-Shield #220: Power down shield when micro-controller loses power

lviss, 2021-02-01:
Hi thanks for this awesome library! I'm using it with a Google Fi data-only sim and it works great.

How can I power down the shield when the micro-controller loses power? Currently if I de-power the Arduino I'm using to control the shield, the shield keeps running from the battery. I don't want to completely discharge the battery repeatedly.

My use-case is an asset tracker wired into the accessory power from a vehicle. The driver won't manually turn off the tracker, and the micro-controller wont know when it is about to lose power to issue `fona.powerDown()`.

I was trying to figure out a way to put a pull-down resistor on the PWRKEY pin, but I'm not sure that's possible.

Thanks again!

aemoreno1982, 2021-02-01:
Can do what generally dash cams do.  Have a large capacitor attached to your power rail.  Then have an interrupt on the Arduino that senses when ignition is off.  When user turns off ignition, Arduino runs off the capacitor and you have just enough time to power down the modem and perform any house cleaning before the Arduino loses power.

thesoaringpotato, 2021-02-01:
Run a large capacitor on your power rail.  Then have an interrupt on your Arduino that ties to the ignition of the vehicle (watch the volts and figure out how to step it down, or you'll fry the Arduino, or better yet, run a relay that grounds the pin when ignition is off).  When the user turns off ignition, Arduino continues to be powered by capacitor and senses ignition is now off.  Have Arduino shut off modem and perform any other house cleaning.  Arduino will then power off when capacitor runs out.

lviss, 2021-02-01:
That seems like it'll work, but the packaging wont be as nice. Any other ideas?

Maybe a future version of the shield could add an option to power down when power goes away on some pin.

lviss, 2021-02-01:
How about this:

To turn the shield off, you hold down the power button for some amount of time (1.2s for SIM7000). I tested, and it seemed I could hold down the button for quite a while, and when I released it, the shield powered down.

I can accomplish the same thing by `digitalWrite(FONA_PWRKEY, LOW);`, and then whenever the MCU loses power it's as if the button was released.

Is there an upper limit to how long that PWRKEY pulse can be? I tested for a minute or two. Are there any downsides to this?

botletics, 2021-02-02:
Per Table 9 of the SIM7000 Hardware Design doc, there is no upper limit for the time it takes for PWRKEY to be LOW to power off the module.

lviss, 2021-02-02:
Ok, I'll do it that way and report back if I run into anything unexpected. Thank you!