botletics/SIM7000-LTE-Shield #226: Sim 7000A buffer, Adafruit not receiving data

Alfredfoby, 2021-03-22:
HARDWARE: Arduino MEGA, SIM 7000A, Hardware Serial 1

So I am still having the issue where my data stops getting to Adafruit IO. I seem to think it has something to do with the buffer getting full on the SIM 7000A, Is there a way to clear this buffer in arduino code? I think this is the cause of the issue because if you reset the system it starts transmitting data. So I have a few questions.

1). How do i reset the buffer using arduino code or changing the library slightly?
2). How can i reset the modem in arduino code and connect it back to the arduino SPI interface? because when i reset using code it never enables gprs or get any connection.  See below for the code i used to reset the modem. Please let me know what I am doing wrong.

```
void modemReset() {
  digitalWrite(FONA_RST, LOW);
  digitalWrite(FONA_PWRKEY, LOW);
  vTaskDelay(400);
  digitalWrite(FONA_RST, HIGH);
  digitalWrite(FONA_PWRKEY, HIGH);

  fonaSerial -> begin(115200);
  //fonaSS.println("AT+IPR=9600");//configure baud rate
  //fonaSS.begin(9600);

  uint8_t imeiLen = fona.getIMEI(imei);
  Serial.print("Module IMEI: ");
  Serial.println(imei);
  fona.setFunctionality(1);
  fona.setNetworkSettings(F("hologram"));
  if (!fona.enableGPRS(false)) {
    Serial.println(F("Failed to disable GPRS!"));
  }

  while (!fona.enableGPRS(true)) {
    Serial.println(F("Failed to enable GPRS, retrying..."));
    vTaskDelay(300);
  }
}
```
Thanks in Advance

Alfredfoby, 2021-03-22:
Okay, Thank you I will try that. Going back to one of my questions, is there a way to clear the buffer and increase the buffer size?

botletics, 2021-03-22:
You can change the buffer size by increasing the number in the line `char replybuffer[255];` and if you need to clear the entire array you can use `memset(0, replybuffer, sizeof(replybuffer));`. Since you're using an Arduino Mega, you should be able to increase the buffer size without memory issues.

Alfredfoby, 2021-03-23:
Just for clarification, Using the `memset(0, replybuffer, sizeof(replybuffer));` cannot be done in arduino code but rather in the library, Adafruit_Fona. Im not necessarily sure.

Thank you.

botletics, 2021-03-23:
It can be done in Arduino code.

Alfredfoby, 2021-03-24:
Sorry if Im asking too many naive questions but how would you do it through arduino code? Considering the variable was declared in the library.

botletics, 2021-03-24:
Sorry, I'm not sure what you're asking. You can use that line directly in your Arduino code since it's built into the Arduino IDE for AVR.

Alfredfoby, 2021-03-24:
Thank you, Essentially what I am asking is that the replybuffer variable is in the library so if i declare a char array `char replybuffer[255]` the variable replybuffer would always be empty because it is not is the actual buffer in the library. I dont know if that makes more sense.

botletics, 2021-03-24:
If you declare a char array like `char replybuffer[255]` it's empty at the start, but when the code runs you're going to be filling it. So to forcefully empty it, you can use memset().

Alfredfoby, 2021-05-13:
It now works

closed 2021-05-13 by Alfredfoby