botletics/SIM7000-LTE-Shield #227: ESP32 LTE Demo will not compile

star297, 2021-03-23:
I get this error trying to compile the ESP32 LTE Demo

C:\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::postData(const char*, uint16_t, const char*, const char*, const char*)':
C:\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:2045:3: error: expected ';' before 'readline'
   readline(10000);
   ^

Any suggestion's?

Many thanks

thesoaringpotato, 2021-03-23:
It's a typo in the library.  
In Adafruit_FONA.cpp, line 2041, add an ';' to the end of sendCheckReply(F("AT+CHTTPSCLSE"), ok_reply, 10000)

botletics, 2021-03-23:
Oops, thanks for catching that! I'll update it.

closed 2021-03-23 by botletics

star297, 2021-03-24:
Thanks Guy's,

star297, 2021-03-24:
This had me going round in circles for a while,

ESP32_LTE_Demo 

There's a typo on Tx, Rx in lines 113 and 118

```fonaSS.begin(115200, SERIAL_8N1, FONA_TX, FONA_RX); // baud rate, protocol, ESP32 RX pin, ESP32 TX pin```
```fonaSS.begin(9600, SERIAL_8N1, FONA_TX, FONA_RX); // Switch to 9600```

should be:
```fonaSS.begin(115200, SERIAL_8N1, FONA_RX, FONA_TX); // baud rate, protocol, ESP32 RX pin, ESP32 TX pin```
```fonaSS.begin(9600, SERIAL_8N1, FONA_RX, FONA_TX); // Switch to 9600```

Also I'm not using

```
// For ESP32 hardware serial
#include <HardwareSerial.h>
HardwareSerial fonaSS(1);
```

Just define fonaSS to ESP32 Serial(1);
```#define fonaSS Serial1```

and not change the baud from 115200 to 9600, works fine at 115200 for the SIM808E, SIM7600E, SIM7000E and SIM7070G.

Thanks again.

botletics, 2021-03-24:
Actually the TX/RX is not a typo, because the MCU_TX will be the FONA_RX, and vice versa.

On the baud rate, it's advisable not to use high baud rates with software serial. However, with hardware serial 115200 is just fine.