botletics/SIM7000-LTE-Shield #228: Failed to enable GPRS

nickhuff17, 2021-03-23:
Running the IOT_Example and it wont enable the GPRS? it was able to acquire gps, i know the hologram card is activated because i used it on another project. got any ideas?

nohe427, 2021-04-15:
I am running LTE_DEMO and am not able to connect to Hologram.io network. It connected once and registered but am no longer connecting. Any thoughts why?

botletics, 2021-04-15:
Have you looked through [this page] (https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network)?

nohe427, 2021-04-19:
@botletics - I run the Enable cellular data command via [G] and for some reason, it responds with `Failed to turn on` - Any thoughts what might be causing that? See full output below:

```
FONA basic test
Initializing....(May take several seconds)
Configuring to 9600 baud
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- SMS Ready
	---> AT
	<--- AT
	---> ATE0
	<--- ATE0
	---> ATE0
	<--- OK
	---> AT+GMR
	<--- Revision:1351B03SIM7000A

OK

	---> AT+CPMS="SM","SM","SM"
	<--- +CPMS: 0,10,0,10,0,10
FONA is OK
Found SIM7000
	---> AT+GSN
	<--- 865235030719328
Module IMEI: 865235030719328
	---> AT+CFUN=1
	<--- OK
	---> AT+CGDCONT=1,"IP","hologram"
	<--- OK
-------------------------------------
[?] Print this menu
[a] Read the ADC; 2.8V max for SIM800/808, 0V-VBAT for SIM7000 shield
[b] Read supply voltage
[C] Read the SIM CCID
[U] Unlock SIM with PIN code
[i] Read signal strength (RSSI)
[n] Get network status
[1] Get network connection info
[c] Make phone Call
[A] Get call status
[h] Hang up phone
[p] Pick up phone
[N] Number of SMS's
[r] Read SMS #
[R] Read all SMS
[d] Delete SMS #
[s] Send SMS
[u] Send USSD
[y] Enable local time stamp (SIM800/808/70X0)
[Y] Enable NTP time sync (SIM800/808/70X0)
[t] Get network time
[G] Enable cellular data
[g] Disable cellular data
[l] Query GSMLOC (2G)
[w] Read webpage
[W] Post to website
[2] Post to dweet.io - 2G / LTE CAT-M / NB-IoT
[3] Post to dweet.io - 3G / 4G LTE
[O] Turn GPS on (SIM808/5320/7XX0)
[o] Turn GPS off (SIM808/5320/7XX0)
[L] Query GPS location (SIM808/5320/7XX0)
[E] Raw NMEA out (SIM808)
[S] Create serial passthru tunnel
-------------------------------------

FONA> G
	---> AT+CIPSHUT
	<--- 
	---> AT+CGATT=1
	<--- 
Failed to turn on
FONA> 
```

I get similar issues trying to activate GPS

botletics, 2021-04-19:
Looks like your module is powering off. Please check to make sure you have a LiPo battery connected to your shield, and make sure it has the right polarity (red wire should be on the left side of the connector when you plug it into the shield).

nohe427, 2021-04-19:
My battery pack looks to only plug in one way. Does this look correct?

[image]

botletics, 2021-04-19:
No, the battery polarity is incorrect. Many vendors do that for some reason, but all the ones from Sparkfun and Adafruit have the correct polarity. You could try to carefully pull the pins out and swap them yourself (being very careful not to short the wires together), or just buy one with proper polarity.

lviss, 2021-04-19:
@botletics is right, my battery came with the leads swapped and I had to switch them. You can see that your black wire is going to the + symbol printed on the board.

nohe427, 2021-04-20:
@botletics @lviss  - Ahh, I see the `+` on the board now. That helps. I swapped the wires as suggested in the first comment and now it all seems to work for now. Thank you! Having trouble with GPS, but that might be due to a lack of sight of the sky

nohe427, 2021-04-20:
@botletics & @lviss - Is there a way to power this without a battery pack?

botletics, 2021-04-20:
Yes, you can supply a strong 4V supply to VBAT, but you have to make sure it can sustain high current spikes of up to 2A. Also, it may be helpful to have a large smoothing capacitor across VBAT and GND to dampen voltage fluctuation.