botletics/SIM7000-LTE-Shield #229: Which Modem should I use advice.

star297, 2021-03-24:
This is not an issue, well it is for me. I need some guidance from those that know.
I'm using a couple of SIM7600E and SIM808 modem's is Europe (Spain).
But I want to use NB-IOT or similar.
These are for farm vehicle tracking and immobilizer and are working, for now.
I use the ESP32 and PUBSUB MQTT library, I need wi-fi when the vehicles are parked up in the barn.
The SIM808 is 2G and will be obsolete by 2025 and I don't think the SIM7600 will have the full NB-IOT capability if any.
 
I just got some SIM7070G modems and hoping to use these.
Are these any good? should I be looking at different models or even move to Ublox or Quectel?

One other function I need is OTA firmware updating for the ESP32, I do this over wi-fi atm, but I would like to be able to use GSM.
Do I use HTTP or FTP?

Any advice would be appreciated.

Paul

botletics, 2021-03-24:
The GPS commands don't work for me on the SIM7070G. Not sure why. I'd personally recommend the SIM7000, but then again, I may be biased.

star297, 2021-03-25:
Thanks Timothy, I'll go with your advice, tried the 7000 MQTT example and works great, thanks for your hard work. 

I have since found this..

https://stackoverflow.com/questions/61857667/sim7080g-module-cant-send-data-over-tcp-while-using-gnss/61884727#61884727

So I don't think there's much point continuing with these.

jwallis, 2021-04-01:
@star297 you are in a fairly rural area and the SIM7000E or SIM7000G is working well for you?

star297, 2021-04-02:
@jwallis Yes it is working fine where I am in North Spain.
I use a ESP32 for my controller.
I'm using the SIM7000E but haven't tried any field tests yet. 
Whereas I have been testing the SIM7600E on 4G LTE and that works everywhere, tested at speed on motorways sending MQTT messages (via TCP using PUBSUB library) with no dropout on a 100km trip.
I don't need huge data so I was looking to use the SIM7000 and hoping the IOT bands would have better propagation and penetration, we'll see.
The built in MQTT works well on the SIM7000, connects, sends a 124 byte json string and disconnects in 5 seconds. Trying to get the subscribe to work atm.

jwallis, 2021-04-04:
I am in Texas, USA, and using a SIM7000A, but I am not using a Botletics shield.  I do not get very good coverage with LTE-M and now I think it's because the antenna design of the board I'm using is not very good.  It works well in the city, but not well outside the city.  It can send SMS, but not receive because it does not get notification of the incoming SMS.

I'm going to try the SIM7500A to see if it is LTE-M coverage, or the antenna design of the board.  The SIM7000 and SIM75000 breakout boards that I'm using are almost exactly the same except for the SIM module.

star297, 2021-04-04:
Antennas make a huge difference. PCB and self adhesive type aren't very good.
I use something similar to this:

https://www.amazon.ca/dp/B07VQPDXLH/ref=sspa_dk_detail_1?pd_rd_i=B07VQPDXLH&pd_rd_w=2j4ry&pf_rd_p=2c17e944-5508-41c9-9e34-6115f0c88f84&pd_rd_wg=EGIko&pf_rd_r=8R1HGWHZZPZBW01F3X3F&pd_rd_r=c405a3a1-42b0-4a33-afe3-6be9c5fda53e&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExSUxJUEtZRTdIRzgwJmVuY3J5cHRlZElkPUEwNjMyODg5MUs5VVMyN09UQUMyUSZlbmNyeXB0ZWRBZElkPUEwNDAyOTUyMlRINVRMWTFaQ0E0USZ3aWRnZXROYW1lPXNwX2RldGFpbCZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU&th=1

Check the antenna gain at the band frequency you use in your location, some wideband antennas may indicate 7db or more gain, but that will be at the center frequency , get to 800/900Mhz and the gain drops a lot.

Mouser electronics do a wide range of antennas with data sheets so you can be more selective and get one that will work better.

jwallis, 2021-04-05:
Thank you, I really appreciate it!  I thought I was using a reasonably good antenna, but I just bought 3 more much longer ones from digiKey to try.  If only data sheets had some standardization... it feels like comparing apples and oranges sometimes, but I get it in general.

Something I have a question about that I wonder if you happen to know is ok, AT&T operates LTE on multiple bands: 
700 MHz: Bands 12/17/29.
850 MHz: Band 5.
1900 MHz: Band 2.
1700 MHz /2100 MHz: Bands 4/66.
2300 MHz: Band 30.

And my antenna says it has 1.5db of gain at 800MHz, but 4db at 1900MHz... well I sure hope it connects to 1900MHz / band 2!  But of course I imagine it's left to the modem to decide what band to connect to.  I assume it can only connect to one band at a time.  I'm curious if the SIM7000 is smart enough to try multiple bands and connect to the "best" one at any given time.

Anyway, thanks again.  I should have looked harder at antenna specs months ago.