botletics/SIM7000-LTE-Shield #230: Things board.io option

nickhuff17, 2021-03-25:
Fixed my other issue (dead 3.7v Lipo Battery :)...anyway i commented out the dweet.io section and uncommented the thingsboard. when i compile the code, i am getting a error on line 480 about #elif without #if and #endif without #if

botletics, 2021-03-26:
Which code, the LTE_Demo? Please paste the exact error here, thanks.

nickhuff17, 2021-03-26:
Arduino: 1.8.13 (Windows 10), Board: "Arduino Uno"

ThingsBoard:443:2: error: #elif without #if

 #elif defined(PROTOCOL_MQTT_AIO)

  ^~~~

ThingsBoard:480:2: error: #elif without #if

 #elif defined(PROTOCOL_MQTT_CLOUDMQTT)

  ^~~~

ThingsBoard:521:2: error: #endif without #if

 #endif

  ^~~~~

exit status 1

#elif without #if

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Now could this mean i commented out to many lines attempting to do the thingsboard post instead of the dweet.io?

nickhuff17, 2021-03-26:
its the IOT_Example program

botletics, 2021-03-27:
You need to comment out lines 419-420 and delete the `*/` and `*/` on lines 427 and 433. It compiles fine for me after doing this.

closed 2021-03-27 by botletics

nickhuff17, 2021-03-28:
Wow, silly mistakes on my part; thank you; how can I contribute to you for your help? additionally, have you ever ran into a situation where once it compiles and you execute the code, the device will register with the towers, find the location then drop the cell service and fail to get back into the network? it runs the look of "network status 0: not registered, failed to connect to cell network". it does this for my hologram card and my t-mobile card..

the LTE_demo works just fine and has no issues on dropping the cell service, i can send and receive texts all day

botletics, 2021-03-29:
Mhmmm I haven't run into that. Maybe double check that the APN you're using is correct.