botletics/SIM7000-LTE-Shield #231: IoT Example Issues.

mduiop, 2021-03-26:
First I tested with the LTE_DEMO to make sure I could connect to data, send a GET header, and receive GPS Coords.  I then ran the IoT_Example using Hologram and sending a GET header.  I think the only thing I changed was the frequency to send the GET header to 5 minutes.

First thing I notice is that the moduleSetup function keeps on line 201 is getting called in a loop over and over.  I say this because (I added debugging output and) the serial monitor keeps repeating "FONA is OK. Found ???"   I found that "type = fona.type();" type had the value of "7" so I changed the switch/case

case SIM7000:
      Serial.println(F("SIM7000")); break;

to 

case 7:
      Serial.println(F("SIM7000")); break; 

(which happens to be the 7th case) and now it prints SIM7000

That stopped it from printing "Found ???" but still this function seems to get called over and over many times. Eventually the next line (202) is reached.

Then it will usually not find GPS or GPRS . . . 
the one time it did actually get to the line of code that sent the header I did receive a HTTP 601 error. 

I am wondering if I am doing anything wrong.  Again, LTE_Demo allowed me to sent the GET header, etc.

Thanks

mduiop, 2021-03-26:
To update I moved to a different physical location, and it is connecting, attempting to send the HTTP Get header to my web server, but I am receiving a 601 error.  If I copy and paste the URL (with the query string) from the terminal (that the script is attempting to send) into a browser's command line then the browser based application will process the data in the query string . . . so there does not seem to be any issues with the web app that is causing the 601 error.  Thanks

mduiop, 2021-03-27:
At this point the following just loops over and over again.  I am new to Arduino and C/C++ but thought that since this function is not called in the main loop it shouldn't repeat over and over.  It's as if it's being reset somewhere.  I am using the Lipo battery, and the green "done" light is illuminated on the board as well as another green light and flashing blue light (LEDs).  Again, I use LTE_Demo first to check I can make a network connection and then run the IoT_Demo and get this:

** SIMCom Module IoT Example ***
Configuring to 9600 baud
Attempting to open comm with ATs
---> AT
<--- OK
---> ATE0
<--- OK
---> ATE0
<--- OK
---> AT+GMR
<--- Revision:1351B03SIM7000A

OK

---> AT+CPMS="SM","SM","SM"
<--- +CPMS: 1,10,1,10,1,10
FONA is OK
Found SIM7000A (American)
---> AT+GSN
<--- 86⸮*** SIMCom Module IoT Example ***
Configuring to 9600 baud
Attempting to open comm with ATs
---> AT
<--- OK
---> ATE0
<--- OK
---> ATE0
<--- OK
---> AT+GMR
<--- Revision:1351B03SIM7000A

OK

---> AT+CPMS="SM","SM","SM"
<--- +CPMS: 1,10,1,10,1,10
FONA is OK
Found SIM7000A (American)
---> AT+GSN
<--- 8⸮*** SIMCom Module IoT Example ***
Configuring to 9600 baud
Attempting to open comm with ATs
---> AT
<--- OK
---> ATE0
<--- OK
---> ATE0
<--- OK
---> AT+GMR
<--- Revision:1351B03SIM7000A

OK

botletics, 2021-03-27:
Looks like you're running out of memory. Try upgrading to an Arduino Mega instead of an Uno (which I'm assuming you're using?)

mduiop, 2021-03-27:
Hi am using the Mega actually. Thanks

botletics, 2021-03-27:
I just tried the unaltered IoT_Example sketch on my Arduino Uno and it worked.

mduiop, 2021-03-27:
UPDATE: I noticed that the IoT_Example sketch was updated on Github a few minutes ago. I used that version, and now it did work.  Posts GPS coords to my PHP script on a web server then emails me the data. I am having a bit of trouble locking onto a GPS signal so it only worked intermittently then obviously wont send data unless it gets coordinates.  That could be my geographic location.  I'll try it on a drive later.  I will probably go for a powered GPS antenna unless you have other suggestions for improving ability to connect to GPS signal, and data network . . . I notice that I either connect to T-Mibile or AT&T via Hologram so maybe uncommenting some of those variables for forcing certain networks would help?  Thanks!

botletics, 2021-03-27:
GPS is completely independent of the network, so your best bet to improving the signal would be an active GPS antenna.

mduiop, 2021-03-27:
Yes, my comment was a little unclear, sorry.  I understand that network issues and GPS issues are two separate things.  I am having a little trouble with both.  I will take your advice for the GPS and use an active antenna.  For the benefit of anyone looking here for help, I will solder a jumper between the two points on the Botletics board required when using a powered antenna.  For signal strength I will research antennas for that purpose, but was wondering if forcing a certain network protocol would improve things by un-commenting some of the variables . . . probably not . . . just a thought.  Thanks again.

botletics, 2021-03-28:
No, I don't think locking any cellular mode or band will help you. If it's a signal issue, your best bet is to get a better antenna.

closed 2021-04-06 by mduiop