botletics/SIM7000-LTE-Shield #233: IoT_Example.ino disconnects and fails to reconnect after successfully sending GET header

mduiop, 2021-03-30:
Hi,

IoT_Example.ino running on an Arduino Mega with the Botletics shield / Hologram SIM connects to GPS, connects to the cellular network, then sends a GET header to my application.  No problem.  After that I will enter a seemingly infinite loop where the monitor will read "Failed to connect to cell network, retrying..." when the next header is attempting to be sent.  Here's the thing . . . : if I reset the board the first GET header (upon reset) is sent no problem, then the sketch hangs up again trying to connect to the network.  I tried this in the country, in the city, and while in a moving vehicle . . . I do not believe it is the network because, like I said, if I reset everything works the first iteration of the loop.  

I am running IoT_Example.ino without having changed any values (latest version) except the URL the GET is sent to.  The only difference I see in the code between the first time it's sent and the subsequent times that fail is that the first time moduleSetup(); is called after fona.powerOn(FONA_PWRKEY); on Line 200.  Subsequently after line 564 fona.powerOn(FONA_PWRKEY); is once again called, but this time moduleSetup(); is not because it is nested in this logic:

  // Only run the initialization again if the module was powered off
  // since it resets back to 115200 baud instead of 4800.
  #ifdef turnOffShield
    moduleSetup();
  #endif

I am not sure if commenting out the logic will fix my problem, but it's the only difference I see,  I have not tried it yet because do not have access to my machine with the compiler on it, plus the comment seems to warn against this re: the baud rate.  Has anyone had this issue or have any ideas?  Not sure if my observation in the code is relevant, but thought I would do some investigative work before posting.  Thanks

mduiop, 2021-03-30:
Just some stabs in the dark here:

Maybe run moduleSetup() no matte what, or . . . 

Maybe at Line 564 instead of this:
  fona.powerOn(FONA_PWRKEY); // Powers on the module if it was off previously

something like this might help?
//only powerOn if not connected 
 if (fona.getNetworkStatus() != 0)
{
      fona.powerOn(FONA_PWRKEY); // Powers on the module if it was off previously
}

mduiop, 2021-03-31:
UPDATE: I commented out this line in IoT_Example.ino (Line# 562):

fona.powerOn(FONA_PWRKEY); // Powers on the module if it was off previously

And, that issue I initially reported is resolved.  It now sends updated location every 10 seconds without losing the connection.  

I figured that since we already connected in the "loop()" we shouldn't need to connect again?  Maybe (and quite possibly) I am missing something, and there is a reason why we may have been disconnected, and should therefore attempt connection again . . . ?  If so, and someone could please advise then I would appreciate that.  Maybe there is a better way to have handled this?

All the best, and kind regards.

botletics, 2021-03-31:
Is it using the command `AT+CGREG?` or `AT=CREG?`. I had made a small error in updating the library to disregard the module versions (like 7000A or G or E, etc.) and in the code, it's used the wrong command for the `getNetworkStatus()` command. It has since been fixed though, if you update the Adafruit_FONA.cpp file.

mduiop, 2021-03-31:
Unfortunately this did not fix the issue that I'm having. The only thing
that seems to work is commenting out the line I referenced above. After
commenting out that line the application was working. This morning I
reverted back to the original version of that sketch and recompiled with
the updated version of your file from last night. The problem came back
after uncommenting the line I was talking about. I saw the fix that you
implemented but I don't think that was the cause of the issue I'm having.
Let me know if you think I really do need that line there or should I
remove the second function call that's not executing because it's nested in
logic. I'm using voice to text right now but all the references I'm making
are in my previous posts. Thanks for your help

botletics, 2021-03-31:
Could you please post a screenshot of the serial monitor?

botletics, 2021-03-31:
I think you're right. I just updated the IoT_Example. Let me know if it works now!

mduiop, 2021-03-31:
Thanks again. I see what you did. Better than my hack of commenting it out.
That should do it.

closed 2021-04-06 by mduiop