botletics/SIM7000-LTE-Shield #234: sim7070g not working

roysG, 2021-04-02:
I have to sim 7070g, i connected the wires, except the reset and pwr-key, cause i did not know where to connect it in the arduino nano.

I ran your example, all the library is compiled, but i get message now found Adafruit.

What do i miss?

closed 2021-04-02 by botletics

roysG, 2021-04-03:
any reason or any comment for closing?

botletics, 2021-04-03:
It looks like you asked a question and then posted an update of how you got it to work.

roysG, 2021-04-03:
No, i posted your code that no working and in rhe sexond code, i poated the
regular one i took from the original library.

Some how in the original library it found the sim but in your library it is
not fins it, any reason?

In addittion, i can send url requrst but not aee the reaponse, do you know
whqt may be the problem?

botletics, 2021-04-03:
Sorry, it's very hard to read code like that, but maybe it has to do with pin definition. The library has been tested with SIM7070G and it works.

roysG, 2021-04-03:
Ok, i bought the sim 7070g, but wheb i connected the wires to arduino, i
got juats the red light power, but the network light is not working, maybe
you know the reason or suggest how to solve it?

rhr407, 2022-04-18:
I receive no response for HTTP Get request for the `http://httpbin.org/get` URL. Following is the output using the AT Command Tester from `https://m2msupport.net`

```

Checking registration status...

AT+CREG?

+CREG: 2,1,"912","3D73",0

OK
The device is registered in home network.

AT+CGREG?

+CGREG: 2,1,"912","3D73",0,"1"

OK
The device is registered in home network.

Device is registered.. 

Check the network APN...

AT+CGNAPN

+CGNAPN: 0,""

OK
Network did not send APN to the device.
Activate the network bearer...

AT+CNACT=0,1

OK

+APP PDP: 0,ACTIVE
Set up the HTTP URL...

AT+SHCONF="URL","httpbin.org"

OK
Set up the HTTP body length...

AT+SHCONF="BODYLEN",1024

OK
Set up the HTTP header length...

AT+SHCONF="HEADERLEN",350

OK
Initiating HTTP connection...

AT+SHCONN

OK
Get the HTTP connection state...

AT+SHSTATE?

+SHSTATE: 1

OK
HTTP connection is successful...

HTTP get request...

AT+SHREQ="http://httpbin.org/get",1

OK
No response received..

```