botletics/SIM7000-LTE-Shield #236: How to disable DEBUG output

star297, 2021-04-02:
How can I stop the serial AT DEBUG output?
It is handy when getting set up then its a pita later on when I want to debug on my own code.

I normally use something like  ```#define   DEBUG    true``` , can this be implemented somewhere ?

star297, 2021-04-02:
I have found a way but would be nice to be able to control this from the main program.

In FONAConfig.h

comment out the ```#define ADAFRUIT_FONA_DEBUG  ``` as shown below

reload the Aduino IDE and compile.

```

/* ADAFRUIT_FONA_DEBUG
 * When defined, will cause extensive debug output on the
 * DebugStream set in the appropriate platform/ header.
 */

//#define ADAFRUIT_FONA_DEBUG  
```

Mark-Wills, 2021-04-02:
No, this is recommended way to do it and it's done in most C programs in
this way.

The reason it is done like this is it allows the debug stuff to be EXCLUDED
from the program if you don't want it, which reduces the size of the
program - no point in compiling that code into the program if you don't
need it

If you did it at run-time, you would need if{} around all the debug code,
but it would still be compiled into your program, wasting space.

Cheers

closed 2021-04-03 by botletics