botletics/SIM7000-LTE-Shield #238: Reset required to reconnect to network after signal lost in IOT_Example sketch

mduiop, 2021-04-06:
With the SIM7000A shield and an Arduino Mega I can run the IOT_Example sketch without modification (except I changed the URL for GET header to point to my app), and send a GET header via HTTP to my custom application.  My GPS coordinates are sent to the web app every 10 seconds . . . no problem.  However, if I lose the data connection (because I drive through a dead spot for example) the sketch will not reconnect.  It just tries over and over until I reset.  Upon reset it will find the network connection immediately, so the signal exists in the area it just can't reconnect after losing connection . . .  Any thoughts are appreciated.

botletics, 2021-04-06:
Do you know what the serial monitor looks like when that happens?

mduiop, 2021-04-06:
If I remember correctly it keeps on looping though:

"Failed to connect to cell network, retrying..."

At line 283, and the beginning of the main loop you have this code:

  while (!netStatus()) {
    Serial.println(F("Failed to connect to cell network, retrying..."));
    delay(2000); // Retry every 2s
  }

Like I said, this works fine unless I lose an already established connection.  If I reset (by disconnecting and re-connecting the serial monitor) then setup() repeats again.  Perhaps because the following two lines are called from setup:

  fona.powerOn(FONA_PWRKEY); // Power on the module
  moduleSetup(); // Establishes first-time serial comm and prints IMEI

Maybe if the above fails to connect a certain number of times then I call powerOn() and moduleSetup() . . . seems like a hack because I do not fully understand what is going on . . . so maybe there is a better solution (if that would even work).

Thanks

botletics, 2021-04-06:
But when the network check fails, does the module simply not respond to the AT commands?

mduiop, 2021-04-06:
Nothing seems to be freezing or hanging up.  I say that because I'll see a new "Failed to connect to cell network, retrying..." message generated every approx. 2 seconds.  If that answers your question?

botletics, 2021-04-06:
But does it show the actual response to the command `AT+CGREG?` when it says it can't connect?

mduiop, 2021-04-07:
Yes, please see below:

Failed to post data, retrying...
Waiting for 10 seconds

	---> AT+CGREG?

	<--- 
Network status 0: Not registered

Failed to connect to cell network, retrying...
	---> AT+CGREG?
	<--- AT+CGREG?
Network status 0: Not registered

Failed to connect to cell network, retrying...
	---> AT+CGREG?
	<--- AT+CGREG?
Network status 0: Not registered
Failed to connect to cell network, retrying...
	---> AT+CGREG?

	<--- AT+CGREG?
Network status 0: Not registered

Failed to connect to cell network, retrying...

botletics, 2021-04-07:
Ahh OK. It's the `moduleSetup()` that's re-establishing the UART connection when it hangs. In that case, you could simply put a counter that increments each time it fails to receive a response, then call `moduleSetup()` when the counter reaches a certain value.