botletics/SIM7000-LTE-Shield #239: max() should be min()

jwallis, 2021-04-08:
@botletics
Sorry to be lazy about this, but it's a 5 second fix for your next update.  In my fork I noticed in
uint8_t Adafruit_FONA::getGPS(uint8_t arg, char *buffer, uint8_t maxbuff) {...

the line
uint8_t len = max(maxbuff-1, (int)strlen(p));

should use min() instead of max() to avoid writing beyond the end of "buffer"

botletics, 2021-04-09:
From looking at it briefly, I think max() is OK because `p` only goes up to 6, and the `len` will only go up to `maxbuff-1`, which will be safe to index.

jwallis, 2021-04-09:
While it's conceivable that len(p) will be > than maxbuff-1, apparently it's not a problem.  I found the commit and it was from 2015.  Either ladyAda was just working too quickly or there's a reason I don't understand.  I have no doubt she's a far better coder than me.  Closing.

closed 2021-04-09 by jwallis