botletics/SIM7000-LTE-Shield #241: Arduino Uno memory requirements for IoT_Example MQTT

DavyJ8675, 2021-04-10:
Is the Arduino Uno capable of running the IoT_Example code for the MQTT when using the stock code? I followed the Youtube tutorial for the MQTT GPS Tracker but the code will not compile as it keeps stating that:

"data section exceeds available space in boardSketch uses 29498 bytes (91%) of program storage space. Maximum is 32256 bytes.

Global variables use 2107 bytes (102%) of dynamic memory, leaving -59 bytes for local variables. Maximum is 2048 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino Uno."

I have attached the libraries in my Arduino folder. Perhaps there is something there that should not be? In a 2018 post I saw someone posted with a similar issue but since there has been code revisions since I wasn't sure if an Uno was no longer suitable for this build and if as MEGA is now required?

[image]

DavyJ8675, 2021-04-10:
I tried deleting some lines but I was conservative as I did not know what could be removed without ruining the flow of the code. This did not reduce the memory usage in the error statement.

botletics, 2021-04-11:
Yea the Adafruit IO part is really hard to optimize because their library is already so memory-intensive and there's not much I can do. The CloudMQTT example can be expanded to other MQTT brokers and takes up much less memory (77% of flash memory for Arduino Uno). Or you could just upgrade to an Arduino Mega which will have plenty of memory for the sketch without any changes in code.

DavyJ8675, 2021-04-11:
I see. Thanks for the fast response. If I may make a suggestion, it could be useful to clarify in the Youtube tutorial's description. This would save a lot of repeat questions as I'm guessing other's have asked this before. I only say this since the video references following the LTE Demo on Instructables which uses the UNO. I'll get my hands on a MEGA. I'm assuming there are no code changes for using the MEGA vs the UNO?

botletics, 2021-04-11:
No code changes for the Mega vs the Uno.

closed 2021-04-11 by DavyJ8675