botletics/SIM7000-LTE-Shield #243: HTTP POST with body not working on SIM7070G

KnightOfNih, 2021-04-13:
Hi Guys,

First off wanted to say thank you for the great library.  It's been a god send for understanding how AT commands work with these modules.

I'm having difficulty getting the provided examples to work when appending a body to an HTTP POST.  Without the body, everything works fine, no problems.  But when I add the body portion, it fails.  I was hoping you may be able to provide insight.

The relevant problem code.  In the logs, this all seems to finish correctly.  On submittal it fails.

```
void addBodyToPost()
{
  // Sending plan text and wait one second, b/c I've seen that in other tutorials
  fona.HTTP_addHeader("Content-Type", "text/plain", 10);
  delay(1000);

  fona.HTTP_addHeader("Host", "b225f0b114f9.ngrok.io", 21);
  delay(1000);

  fona.HTTP_addHeader("User-Agent", "IOE Client", 10);
  delay(1000);

  // Add the SHBOD.  It will return a ">" when it is ready for data
  char cmdBuff[150];
  char body[] = {"foo"};
  int bodySize = sizeof(body) / sizeof(body[0]);

  // Format your command and wait up to 2 seconds for SIM7070 go respond
  sprintf(cmdBuff, "AT+SHBOD=%i,10000", bodySize);
  fona.getReply(cmdBuff, 2000);

  // If you get a reply, hope it's a prompt
  if (strstr(fona.replybuffer, ">") == NULL)
    Serial.println("No > in reply buffer");

  // Send the data to the module and wait for OK
  if (!fona.sendCheckReply(body, "OK", 2000))
    Serial.println("No OK in reply buffer.");

  // Data added!  Ha.  That's what they WANT you to think.
}
```

The entire sketch.

```
#include <Arduino.h>
#include "Adafruit_FONA.h" // https://github.com/botletics/SIM7000-LTE-Shield/tree/master/Code

#define SIMCOM_7070 // I'm using one of these

#define FONA_PWRKEY 18
#define FONA_RST 5
#define FONA_TX 16 // ESP32 hardware serial RX2 (GPIO16)
#define FONA_RX 17 // ESP32 hardware serial TX2 (GPIO17)

// For ESP32 hardware serial
#include <HardwareSerial.h>
HardwareSerial fonaSS(1);

// The fona library.  Excellent!
Adafruit_FONA_LTE fona = Adafruit_FONA_LTE();

void addBodyToPost()
{
  // Sending plan text and wait one second, b/c I've seen that in other tutorials
  fona.HTTP_addHeader("Content-Type", "text/plain", 10);
  delay(1000);

  fona.HTTP_addHeader("Host", "b225f0b114f9.ngrok.io", 21);
  delay(1000);

  fona.HTTP_addHeader("User-Agent", "IOE Client", 10);
  delay(1000);

  // Add the SHBOD.  It will return a ">" when it is ready for data
  char cmdBuff[150];
  char body[] = {"foo"};
  int bodySize = sizeof(body) / sizeof(body[0]);

  // Format your command and wait up to 2 seconds for SIM7070 go respond
  sprintf(cmdBuff, "AT+SHBOD=%i,10000", bodySize);
  fona.getReply(cmdBuff, 2000);

  // If you get a reply, hope it's a prompt
  if (strstr(fona.replybuffer, ">") == NULL)
    Serial.println("No > in reply buffer");

  // Send the data to the module and wait for OK
  if (!fona.sendCheckReply(body, "OK", 2000))
    Serial.println("No OK in reply buffer.");

  // Data added!  Ha.  That's what they WANT you to think.
}

void postRequest()
{

  addBodyToPost();

  // Send your reply and wait up to five seconds for the module to confirm
  fona.sendCheckReply("AT+SHREQ=\"http://b225f0b114f9.ngrok.io\",3", "OK", 5 * 1000);

  // Read responses for up to 10 seconds
  // Note: What's odd is that when adding a body, this fails immediately
  // Parse response status and size
  // Example reply --> "+SHREQ: "POST",200,452"
  uint16_t status, datalen;
  fona.readline(10 * 1000);

  // What did it return?
  DEBUG_PRINT("\t<--- ");
  DEBUG_PRINTLN(fona.replybuffer);

  // Did we get a status reply of 200?
  if (!fona.parseReply(F("+SHREQ: \"POST\""), &status, ',', 1))
  {
    Serial.println("Unable to get status");
    return;
  }

  // The moment of truth
  if (status != 200)
  {
    // D'oh ... what went wrong?
    Serial.println("No soup for you!  Failure ....");
  }
  else
  {
    // Yus!! We got a reply from the server.  Let's see what it says!
    Serial.println("Huzzah!  We have a response ...");

    if (!fona.parseReply(F("+SHREQ: \"POST\""), &datalen, ',', 2))
    {
      Serial.println("Unable to get data");
    }

    DEBUG_PRINT("HTTP status: ");
    DEBUG_PRINTLN(status);
    DEBUG_PRINT("Data length: ");
    DEBUG_PRINTLN(datalen);

    // Read server response
    fona.getReply(F("AT+SHREAD=0,"), datalen, 10000);
    fona.readline();
    DEBUG_PRINT("\t<--- ");
    DEBUG_PRINTLN(fona.replybuffer); // +SHREAD: <datalen>
    fona.readline(10000);
    DEBUG_PRINT("\t<--- ");
    DEBUG_PRINTLN(fona.replybuffer); // Print out server reply
  }
}

void setup()
{

  pinMode(FONA_RST, OUTPUT);
  digitalWrite(FONA_RST, HIGH); // Default state

  fona.powerOn(FONA_PWRKEY); // Power on the module

  Serial.begin(9600);
  Serial.println(F("ESP32 SIMCom Basic Test"));
  Serial.println(F("Initializing....(May take several seconds)"));

  fonaSS.begin(115200, SERIAL_8N1, FONA_TX, FONA_RX); // baud rate, protocol, ESP32 RX pin, ESP32 TX pin

  Serial.println(F("Configuring to 9600 baud"));
  fonaSS.println("AT+IPR=9600");                    // Set baud rate
  delay(100);                                       // Short pause to let the command run
  fonaSS.begin(9600, SERIAL_8N1, FONA_TX, FONA_RX); // Switch to 9600
  if (!fona.begin(fonaSS))
  {
    Serial.println(F("Couldn't find FONA"));
    while (1)
      ; // Don't proceed if it couldn't find the device
  }

  uint8_t type = fona.type();
  Serial.println(F("FONA is OK"));
  Serial.print(F("Found "));
  switch (type)
  {
  case SIM7070:
    Serial.println(F("SIM7070"));
    break;
  }

  // Print module IMEI number.
  char imei[16] = {0};
  uint8_t imeiLen = fona.getIMEI(imei);
  if (imeiLen > 0)
  {
    Serial.print("Module IMEI: ");
    Serial.println(imei);
  }

  // Set modem to full functionality
  fona.setFunctionality(1);               // AT+CFUN=1
  fona.setNetworkSettings(F("hologram")); // For Hologram SIM card
  fona.setPreferredMode(38);              // Use LTE only, not 2G
  fona.setPreferredLTEMode(1);            // Use LTE CAT-M only, not NB-IoT

  int attemptCount = 0;
  while (!fona.getNetworkInfo())
  {
    delay(1000);
    attemptCount++;
    if (attemptCount == 10)
    {
      ESP.restart();
    }
  }

  // Attempt to turn on data
  attemptCount = 0;
  while (!fona.enableGPRS(true))
  {
    delay(1000);
    attemptCount++;
    if (attemptCount == 10)
    {
      ESP.restart();
    }
  }

  char url[] = {"http://b225f0b114f9.ngrok.io"};
  fona.HTTP_connect(url);

  postRequest();

  fona.getReply("AT+SHDISC", 10000); // Disconnect HTTP

  delay(10 * 1000);
  ESP.restart();
}

void loop(){};
```

The console output

_ESP32 SIMCom Basic Test
Initializing....(May take several seconds)
Configuring to 9600 baud
Attempting to open comm with ATs
        ---> AT
        <--- OK
        ---> ATE0
        <--- OK
        ---> ATE0
        <--- OK
        ---> AT+GMR
        <--- Revision:1951B03SIM7070

OK

        ---> AT+CPMS="SM","SM","SM"
        <--- +CPMS: 0,10,0,10,0,10
FONA is OK
Found SIM7070
        ---> AT+GSN
        <--- 869777041783823
Module IMEI: 869777041783823
        ---> AT+CFUN=1
        <--- OK
        ---> AT+CGDCONT=1,"IP","hologram"
        <--- OK
        ---> AT+CNMP=38
        <--- OK
        ---> AT+CMNB=1
        <--- OK
        ---> AT+CPSI?
        <--- +CPSI: LTE CAT-M1,Online,310-410,0x3A85,46161679,402,EUTRAN-BAND12,5110,3,3,-16,-81,-51,8
OK replyOK
        ---> AT+CNACT=0,1
        <--- OK
        <--- +APP PDP: 0,ACTIVE
        ---> AT+CNACT?
        <--- +CNACT: 0,1,"10.155.118.146"
        ---> AT+SHCONF="URL","http://b225f0b114f9.ngrok.io"
        <--- OK
        ---> AT+SHCONF="BODYLEN",1024
        <--- OK
        ---> AT+SHCONF="HEADERLEN",350
        <--- OK
        ---> AT+SHCONN
        <--- OK
        ---> AT+SHSTATE?
        <--- +SHSTATE: 1
        ---> AT+SHCHEAD
        <--- OK
        ---> AT+SHAHEAD="Content-Type","text/plain"
        <--- OK
        ---> AT+SHAHEAD="Host","b225f0b114f9.ngrok.io"
        <--- OK
        ---> AT+SHAHEAD="User-Agent","IOE Client"
        <--- OK
        ---> AT+SHBOD=4,10000
        <--- > 
        ---> foo
        <--- OK
        ---> AT+SHREQ="http://b225f0b114f9.ngrok.io",3
        <--- OK
        <--- *PSUTTZ: 21/04/13,16:21:50","-20",1
Unable to get status
        ---> AT+SHDISC
        <--- AT+SHDISC_

Am I just missing something obvious?  I've spent a couple of days on this adding/removing headers and modifying the SHBOD length, delay timers, etc to no avail.

Any help would be greatly appreciated.  Thank you for the awesome library.

botletics, 2021-04-13:
I ran into the same issue with SIM7070G, where it errors out after adding the JSON body. I think it's a SIMCom firmware issue.

KnightOfNih, 2021-04-13:
Thank you for the quick response.  Did you run into the same issue when you tried with a SIM7080?  Or was this particular to the 7070?

botletics, 2021-04-13:
I have not tried the SIM7080.

akshayb302, 2021-05-15:
Hi guys,
I am using SIM7070G and I am getting **AT+SHCONN: ERROR**
I have followed the step by step procedure shown in the HTTP Application Note. 
But I am still getting that error. Can anyone help?
[image]

KnightOfNih, 2021-06-10:
Just wanted to loop back and close out this ticket.  I was able to get HTTP (no HTTPS sadly) to work with url form encoded so that's one way to skin the cat.  AT commands below.  

```
// Beginning of the program

// Ping the cell module until it responds to commands

        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- SMS Ready
        ---> AT
        <--- AT

// Turn off any echos.  Dealing with them only causes headaches

        ---> ATE0
        <--- ATE0
        ---> ATE0
        <--- OK

// Turn off the clock so we don't get any unsolicited timestamp info

        ---> AT+CSCLK=0
        <--- OK
        ---> AT+GMR
        <--- Revision:1951B05SIM7070
		<--- OK
        ---> AT+CPMS="SM","SM","SM"
        <--- +CPMS: 0,10,0,10,0,10

// Not a real GSN number
        ---> AT+GSN
        <--- 861340049499080

// Not my real CCID, although the f at the end appears to be a quirk and is normal

        ---> AT+CCID
        <--- 8942502301314194715f

// Check signal quality.  I continue to do this until I get something other than a 99

        ---> AT+CSQ
        <--- +CSQ: 99,99
        <--- OK
        ---> AT+CSQ
        <--- +CSQ: 24,99
        <--- OK

// Signal quality looks good.  Let's check if we are registered to a carrier.  Sometimes this can take a while so I'll repeat this command every 2 seconds.  Note that if you have never registered to a network, this can take a minute or two.  I'm using Hologram.io and even after I registered my SIM, it still took overnight for the SIM to register to a carrier successfully

        ---> AT+CPSI?
        <--- +CPSI: LTE CAT-M1,Online,310-260,0x8201,15791127,64,EUTRAN-BAND12,5035,2,2,-17,-96,-66,9
        ---> AT+CGDCONT=1,"IP","hologram"
        <--- OK

// This was important for my SIM7070G.  Full functionality, LET only
        ---> AT+CFUN=1
        <--- OK
        ---> AT+CNMP=38
        <--- OK
        ---> AT+CMNB=1
        <--- OK
        ---> AT+CLTS=0
        <--- OK

// The ordering of the AT commands from here on out is especially important

        ---> AT+CNACT=0,1
        <--- OK
        <--- +APP PDP: 0,ACTIVE
        ---> AT+CNACT?
        <--- +CNACT: 0,1,"10.99.150.52"

// This is a random URL setup to forward to my backend server.  Highly recommend ngrok.io

        ---> AT+SHCONF="URL","http://5fb2dbe54603.ngrok.io"
        <--- OK

// Set these to max which is 1024 and 350

        ---> AT+SHCONF="BODYLEN",1024
        <--- OK
        ---> AT+SHCONF="HEADERLEN",350
        <--- OK
        ---> AT+SHCONN
        <--- OK
        ---> AT+SHSTATE?
        <--- +SHSTATE: 1

// Add some parameters

        ---> AT+SHPARA=fruit,apple
        <--- OK
        ---> AT+SHPARA=count,6
        <--- OK
        ---> AT+SHREQ=http://5fb2dbe54603.ngrok.io,3
        <--- OK
        <--- +SHREQ: "POST",200,9

// Wait a bit, hoping for a response

        ---> AT+SHREAD=0,9
        <--- OK
        <--- +SHREAD: 9

// data sent back from my server.  
        <--- It works!
```

closed 2021-06-10 by KnightOfNih