botletics/SIM7000-LTE-Shield #244: couldnt find fona

sagarchowdhury, 2021-04-24:
Hi timothy ,

I bought the bioletics SIM7000 LTE/NB-IoT shield for Arduino and i am using hologram sim card i am getting the same error couldnt find fona

FONA basic test
22:33:03.064 -> Initializing....(May take several seconds)
22:33:03.064 -> Configuring to 9600 baud
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
22:33:05.340 -> 	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
Timeout: No response to AT... last ditch attempt.
22:33:25.795 -> 	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> ATE0
	<--- 
	---> ATE0
	<--- 
22:33:28.961 -> Couldn't find FONA

this is the output of the serial monitor

botletics, 2021-04-24:
Hi there, do you see the green PWR LED lighting up on the shield? Also, do you have a 3.7V LiPo battery connected (with red wire on the left side of the connector when you plug it into the board)?

sagarchowdhury, 2021-04-24:
> Hi there, do you see the green PWR LED lighting up on the shield? Also, do you have a 3.7V LiPo battery connected (with red wire on the left side of the connector when you plug it into the board)?

Yes the green power led is on ,and i have used 3.7v lipo battery 2000 mah ,and i have connected the red wire on the ledt side of the connector as well. I haven't soldered the heads instead i have used breadboard to plugin the headers

botletics, 2021-04-24:
Make sure your connections on 5V, GND, and pins 10-11 are secure to the Arduino.

sagarchowdhury, 2021-04-24:
> Make sure your connections on 5V, GND, and pins 10-11 are secure to the Arduino.

I did connect it like a shield with the help of jumper wires ,just a quick question ,if i am doing external wiring pin 10 and 11 of the sim7000a shield should be connected to the Rx and Tx of the pin of the arduino uno that is pin 0 and 1 of the arduino uno board

botletics, 2021-04-24:
No, pins 10-11 on the shield should be connected to pins 10-11 on the Arduino, as if it were directly plugged into the Arduino.

sagarchowdhury, 2021-04-25:
> No, pins 10-11 on the shield should be connected to pins 10-11 on the Arduino, as if it were directly plugged into the Arduino.

I have connected 10-11 pins of the shield with the Arduino and also connected the 5v and gnd pins of the Arduino with the shield ,i havent soldered the headers yet ,i am using jumper wires instead ,do you think it may be a loose connection

botletics, 2021-04-25:
If the green "PWR" LED is on (not the green "DONE" LED) and there is a secure connection on pins 10-11 with a proper LiPo battery connected, it should work.

LivingSensor, 2021-09-22:
What would be the next step to trouble shoot this situation? I have the headers soldered onto the shield and attached the whole shield to the Arduino. I'm still getting:
"
Timeout: No response to AT... last ditch attempt.
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> ATE0
	<--- 
	---> ATE0
	<--- 
Couldn't find FONA
"

botletics, 2021-09-26:
Do you have a LiPo battery plugged in? (And check polarity: red wire should be on left side of the connector when you plug it into the shield).

LivingSensor, 2021-09-28:
Yes. I have a LiPo battery plugged in as well and the red wire is located on the left of the connector.

[image]

botletics, 2021-09-29:
Are you able to turn on the SIM7000? You should see the other green "PWR" LED light up. You can press and hold the pushbutton for a couple seconds and the green LED should light up shortly thereafter.

howdoyoudoit, 2021-10-01:
excuse me if i got the same error but the wiring is SIM7000 and ESP8266. But i did not use the lipo batery instead i'm using the direct wiring between the two device.

is it because it have connected to lipo battery also?
 Thanks

botletics, 2021-10-01:
If you are using the Botletics shield, yes, you need a LiPo battery.

ss4adam, 2022-01-22:
Timothy, I also bought a brand new DF robot SIM7000A. Located in the USA. I am using an Arduino Uno the Ardunio USB for power (the same setup as the video on youtube where you use GPS in a car). I have the exact same problem as listed in this post. I followed the instructions exactly as shown in this GitHub wiki
[image]
.