botletics/SIM7000-LTE-Shield #246: Botletics SIM7000A shield and Arduino DUE issues

rhino171, 2021-04-26:
I have Arduino DUE and SIM7000A. I have already soldered the pins/headers so that the SIM7000A shield goes directly on top of the Due.

I have also already downloaded the libraries. I am using hardware serial from the example codes because software serial won't work for the Due.
After I change the codes to fit the Due, it compiles and uploads, and on the serial monitor, I get 

Couldn't find SIM7000
FONA is OK
Found SIM7500

Interestingly, I get the same results when SIM7000A is not even connected. So, I'm guessing that Arduino Due and SIM7000A shield are not even connected even when the shield is directly on top of the Due.

Next, I also tried changing the RX/TX pins to all fit all Serial ports of Due (Serial, Serial1, Serial2, Serial3) but no luck there as well.

How can I fix this issue? Is there something wrong I'm doing?

Please help me. Thank you!

botletics, 2021-04-26:
Please follow the wiki's pinouts page regarding 3.3V microcontrollers.

rhino171, 2021-04-26:
I'm sorry but I am new to this. Where would I find that?

botletics, 2021-04-26:
You can find it [here](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Pinouts#33v-arduino-form-factor-boards)