botletics/SIM7000-LTE-Shield #247: AWS IoT MQTT using SIM7000A Shield

rhino171, 2021-04-27:
Hello, I want to use Arduino Due and SIM7000A to send and receive data from/to AWS IoT Core as MQTT. Are there any examples I can look at? So subscribe/publish to a topic. Any help? Thank you!

davegravy, 2021-05-21:
> Hello, I want to use Arduino Due and SIM7000A to send and receive data from/to AWS IoT Core as MQTT. Are there any examples I can look at? So subscribe/publish to a topic. Any help? Thank you!

Also looking for this

tomlankhorst, 2021-05-25:
https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/AT%20Command%20Logs/SIM7000_AWS_Log.txt

Looking for an example that only validates server against CA though. 
So no client certs (client auth via username/password).

Anyone achieved this?

botletics, 2021-05-26:
Please see the `addRootCA()` and `TCPconnect()` functions [here](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Library-Functions#fonaaddrootca). Please also see [this set of AT commands](https://github.com/botletics/SIM7000-LTE-Shield/issues/165#issue-568893959) that another user tried that worked for him.

davegravy, 2021-06-03:
@tomlankhorst I achieved this recently. See https://github.com/botletics/SIM7000-LTE-Shield/issues/58#issuecomment-851797623