botletics/SIM7000-LTE-Shield #248: MQTT Connection Fail

najabhaider96, 2021-04-27:
Hi,

I'm using SIM7000C and successfully run HTTP request on NB-IoT now I'm trying to connect with MQTT broker.
I have 2 MQTT Brokers one is IBM Watson IoT Platform and second one is CloudMQTT.
In both brokers I'm failing every time to connect MQTT connection on **AT+SMCONN** command.
All commands runs successfully according to response but on this command It always return **ERROR**.

**Here is my commands with responses.
AT+CSQ
+CSQ: 27,99

OK

AT+CGDCONT=1,"IP","nbiot"  
OK

AT+CNACT=1,"nbiot"
OK

AT+CNACT?
+CNACT: 1,"xx.xx.xx.xx"

OK

AT+SMCONF="URL","HOST:PORT" 
OK

AT+SMCONF="CLIENTID","clientname" 
OK

AT+SMCONF="USERNAME", "username"
OK

AT+SMCONF="PASSWORD", "password"
OK

AT+SMCONN // This always return ERROR in both MQTT brokers.
ERROR**

Please help me to resolve this issue or if any suggestions please tell me.

Thank You

najabhaider96, 2021-04-28:
Hi,

When I used this **https://www.emqx.io/mqtt/public-mqtt5-broker** public MQTT Broker with these set of commands without any Username and Password it worked for me.
But I can't use this in my production I don't have any option to use any broker without Username and Password authentication in my production. :(

Here is my commands with responses.
AT+SMSTATE?
+SMSTATE: 0
OK

AT+CSQ
+CSQ: 22,99

OK

AT+CGDCONT=1,"IP","APN" 
OK

AT+CNACT=1,"APN"
OK

+APP PDP: ACTIVE

AT+SMCONF="URL","broker.emqx.io:1883" 
OK

AT+SMCONF="CLIENTID","Device001" 
OK

AT+SMCONF?
+SMCONF 

CLIENTID: "Device001"
URL: broker.emqx.io:1883
KEEPTIME: 60
USERNAME: ""
PASSWORD: ""
CLEANSS: 0
QOS: 0
TOPIC: ""
MESSAGE: ""
RETAIN: 0

OK

AT+SMCONN
OK

AT+SMSUB="update",1
OK

AT+SMPUB="update","5",1,1
> hello

OK

+SMSUB: "update","hello"

AT+SMPUB="update","8",1,1
> HI Najaf

OK

+SMSUB: "update","HI Najaf"

jatayu86, 2022-12-27:
@najabhaider96 i face the similar issue, did you find a resolution for this?

psiborg99, 2024-03-27:
15:04:28.697 -> AT+SMCONF="URL","(mqtt broker url here):1883"
15:04:28.739 -> ERROR

I wrote an code for sending data to mqtt but it was not working now i tried doing it with at commands still it throws error i am using SIM800L and rp2040 as host