botletics/SIM7000-LTE-Shield #251: +CME ERROR: operation not allowed

amitc1106, 2021-05-25:
I am not able to connect , on running I am getting following on serial monitor, 

Attempting to open comm with ATs
	---> AT
	<--- ERROR
	---> AT
	<--- OK
	---> AT
	<--- OK
	---> ATE0
	<--- OK
	---> ATE0
	<--- OK
	---> AT+GMR
	<--- Revision:1351B03SIM7000A

OK

	---> AT+CPMS="SM","SM","SM"
	<--- +CPMS: 3,10,3,10,3,10
FONA is OK
Found 	---> AT+GSN
	<--- 865235030116418
Module IMEI: 865235030116418
	---> AT+CFUN=1
	<--- OK
	---> AT+CGDCONT=1,"IP","hologram"
	<--- OK
	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+SAPBR=0,1
	<--- OK
	---> AT+CGATT=0
	<--- OK
	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+CGATT=1
	<--- OK
	---> AT+SAPBR=3,1,"CONTYPE","GPRS"
	<--- OK
	---> AT+SAPBR=3,1,"APN","hologram"
	<--- OK
	---> AT+CSTT="hologram"
	<--- OK
	---> AT+SAPBR=1,1
	<--- OK
	---> AT+CIICR
	<--- OK
Enabled data!
Card initialized. 
	---> AT+CAOPEN?
	<--- OK
Connecting to MQTT... 
	---> AT+CMEE=2
	<--- OK
	---> AT+CMEE?
	<--- +CMEE: 2
	---> AT+CACID=1
	<--- +CME ERROR: operation not allowed
Connection failed

botletics, 2021-05-26:
Are you running the LTE_Demo sketch with mods?

amitc1106, 2021-05-26:
I am making a data acquisition system for field application. It's not a demo sketch. I run a demo sketch as well for IO however it gives me an error in reading GPS data.

amitc1106, 2021-07-20:
> Are you running the LTE_Demo sketch with mods?

I run LTE demo sketch, it works fine. However when I run IOT example, We get this error

botletics, 2021-07-20:
Have you tried posting data via HTTP with the 'G' and '2' options from the menu, just to make sure cellular data on your SIM card works? What SIM are you using?

amitc1106, 2021-07-21:
I tried posting data with "G" and '2', and I could see it is posted on dweet.io 
I am using Hologram sim .

botletics, 2021-07-21:
So it looks like you have `#define SSL_FONA 1` in the .h file. Change that to 0 and see if it posts.

amitc1106, 2021-07-21:
Sorry for my limited knowledge.  which .h file I should look for.

botletics, 2021-07-21:
It's the Adafruit_FONA.h file in the "Code" folder.

amitc1106, 2021-07-22:
Thanks, it resolved the issue.

closed 2021-07-22 by amitc1106