botletics/SIM7000-LTE-Shield #253: Power consumption higher than expected

davegravy, 2021-06-08:
I've tried sleep and eDRX modes and I can't seem to get less than 11mA current draw. From the docs I should be able to get ~2mA. I don't have USB connected, nor GPS enabled. Network LED is disabled.

I'm measuring current using the readout on my GW INSTEK GPD-3303S bench supply which is supplying 4.2V to VBAT. The [specs](https://www.gwinstek.com/en-global/products/downloadSeriesSpec/1476) show it has 0.3% accuracy (if I'm reading that right), so I don't **think** it's measurement error. 

Any idea what I'm doing wrong?

```
---> AT+CSCLK=1
<--- OK
---> AT+CEDRXS=1,4,"0010"
<--- OK
```

davegravy, 2021-06-09:
Problem was I wasn't setting DTR high, I'm now getting expected levels. I think the Wiki section on Power Consumption could make it a bit more clear that DTR is required to enter sleep mode, as it outlines the steps to get there except for the DTR pin.

botletics, 2021-06-09:
Thanks, I added that to the wiki.

closed 2021-06-09 by botletics