botletics/SIM7000-LTE-Shield #257: HTTPS for 7000g

TimoWielink, 2021-07-05:
Searching around it looks like the SIM7000G supports HTTPS, I was not able to get any of the `fona.HTTP_POST()` methods to work so switched to `fona.postData()`

with fona.postData() I was able to post to my endpoint with no problem over HTTP, I don't see any of the HTTPS methods for the SIM7000G

Is it possible to get HTTPS? and are there any methods I can modify? 

`fona.postData("https://test.com", 443, "HTTPS", URL, body)`

@botletics If you have any info on HTTPS for the SIM7000G, happy to learn more

Using V6

johnedstone, 2022-01-10:
Yes, I would be interested in updates to postData so that the SIM7000 could use HTTPS.  I am trying to migrate this [sketch - link](https://github.com/johnedstone/mkrgsm1400-post-json-ssl/blob/main/GsmWebClient_hourly_SSL_Post/GsmWebClient_hourly_SSL_Post.ino) to the botletics SIM7000-LTE-Shield.
BTW: thanks for this great board, great documentation, and great examples!

botletics, 2022-01-10:
Please see [this library function](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Library-Functions#fonaaddrootca) and also [this post](https://github.com/botletics/SIM7000-LTE-Shield/issues/58#issuecomment-919259195) in which someone got SSL working.

johnedstone, 2022-01-10:
Thank you, I will check it out

-------------------------
Ed Johnstone    Ô¿Ô¬
cell: 317.331.1848
email: ***@***.***
*'Discouragement is never from God, as it clouds faith and hope' - St
Ignatius*

*'Be the first to love' - Chiara Lubich*

botletics, 2022-09-08:
Hey guys, figured out how to connect with SSL without verifying certs on the SIM7000! Please do the following:
- Set "SSL_FONA" to 1 in the .h file (remember to save the file before closing)
- Download the latest .cpp file from GitHub
- Open the unedited LTE_Demo example sketch and change "http://dweet.io" to "https://dweet.io" on line 1035 (under the '2' option)
- Upload to the Arduino
- Run the 'G' command to enable data connection, then run '2' to send data to dweet.io using SSL. And voila! 😊

closed 2022-09-08 by botletics