botletics/SIM7000-LTE-Shield #258: Arduino nano

ammarqs, 2021-07-20:
Hello Timothy,

We worked with your Sim7000 modules with the arduino mega, added someone of our code to interface with our servers and everything went well.
As soon as i switched the hardware to the nano board, the HTTP posts stopped working, it keeps saying "Failed to post data, retrying...", and then after a while it just stops trying. I switch to my mega board with the same hardware and everything runs smoothly. 
A few changes to note, i am using the software serial on the nano, on the mega, i used the serial1 line. we commented your software serial lines for fonass, didnt see it being used anywhere. 
Can you point us to a direction to which we can start troubleshooting this. Thanks. 
Also something we noted on the nano is that we get a "Low memory available, stability problems may occur" warning. 

Let me know. thanks.

botletics, 2021-07-20:
Yup, low memory with the Arduino Nano is the major suspect here. If you're going for a compact form factor, there are other alternatives that have much more memory, like SAMD21 Cortex M0 boards (you can brows through Adafruit's "Feather" line of products to see)

ammarqs, 2021-07-20:
yes you are right. 
I am will be looking into a custom build atmega2560 board here, instead of using the mega2560 footprint. 
Thank you.

ammarqs, 2021-07-20:
Well i tried redoing our server side and reducing the message being sent via the post request, and that fixed the issue.
The compile still says "Low memory available, stability problems may occur", but if i control my message structure being sent, it should have no issues. 
any other issues that would arise other than this because of memory?

botletics, 2021-07-20:
It may work, but you just have to keep in mind your goals and think about what you might add to the code later.