botletics/SIM7000-LTE-Shield #268: Ok, now it detects the port, I was trying to use the AT port instead of Diagnostics port

jugkrit, 2021-09-28:
Ok, now it detects the port, I was trying to use the AT port instead of Diagnostics port

_Originally posted by @JGLJGL in https://github.com/botletics/SIM7000-LTE-Shield/issues/156#issuecomment-583055603_

jugkrit, 2021-09-28:
> Ok, now it detects the port, I was trying to use the AT port instead of Diagnostics port

Hi, How did you change

botletics, 2021-10-01:
Use a serial terminal like PuTTY to open the "SimTech HS-USB AT Port 9001 (COM 18)" port in your picture above to send AT commands to it.

closed 2021-10-01 by botletics

jugkrit, 2021-10-06:
I used Putty to send command with SIM7000 already, what command for switch port to AT command in QPST?
I would like to import CA Root Cert to SIM7000, could you suggest.

botletics, 2021-10-09:
Use the EFS Explorer tool. Please see [this post](https://github.com/botletics/SIM7000-LTE-Shield/issues/156)