botletics/SIM7000-LTE-Shield #270: JST connector on SIM7000G

howdoyoudoit, 2021-10-11:
Hello Timothy, I got an issue about the connector on the SIM7000. I have already purchased 3 battery lipo 3.7v with JST connector but it does'nt work out. The connector is not compatible with the connector on SIM7000. Right now i'm clueless how to insert the lipo battery when the connector does not match. As for the second issue, I want to create the contenna GPS tracking system. But the coding you create is for the GPS only. If i want to add device such as magnetic sensor and load cell sensor, how do i want to add the data from the sensors into the IOT?. Do i have to insert the coding of the sensors into the IoT_sketch?. Can you help me in this matter?

Here's the battery that i've purchased and the JST connector also. Can you please have a look?

[SIM7000-connector.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7319877/SIM7000-connector.pdf)
[Battery-Lipo-JST-3.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7319878/Battery-Lipo-JST-3.pdf)
[Battery-Lipo-JST-2.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7319879/Battery-Lipo-JST-2.pdf)
[Battery-Lipo-JST.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7319880/Battery-Lipo-JST.pdf)
[JST-Connectror.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7319881/JST-Connectror.pdf)

howdoyoudoit, 2021-10-12:
Hi Timothy, I have connect the Battery Lipo with the SIM7000 but it seem the data show 'could'nt not find FONA'. Can you please help me?
⸮FONA basic test
Initializing....(May take several seconds)
Configuring to 9600 baud
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
Timeout: No response to AT... last ditch attempt.
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> ATE0
	<--- 
	---> ATE0
	<--- 
Couldn't find FONA

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01a0
3fffff80:  3ffee930 3ffee7fc 3ffeeac8 40201335  
3fffff90:  feefeffe feefeffe feefeffe 3ffeeb30  
3fffffa0:  3fffdad0 00000000 3ffeeb1c 40206568  
3fffffb0:  feefeffe feefeffe 3ffe85d8 401011e1  
<<<stack<<<

Also, if the battery lipo is connected  the SIM7000 is On LED right? but the data shows error.

Here is the pictures of SIM7000 of mine connected with lipo battery with using the usb cables of esp8266 and vice versa.

[without USB Cables.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7329387/without.USB.Cables.pdf)
[Connected-USB-Cables.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/7329390/Connected-USB-Cables.pdf)

botletics, 2021-10-14:
Your LiPo battery polarity is reversed. Please make sure the red wire is on the left side of the connector when you plug it into the shield. The batteries from Sparkfun and Adafruit are all wired correctly, but many vendors swap it.