botletics/SIM7000-LTE-Shield #280: Firmware Upgrade - shield is dead

johnedstone, 2022-01-13:
Yes, I have the LiPo 3.7 V 250mAh batter connected.
Yes, I did not have it connected to a board such as Arduino Uno rev3

Upgraded firmware to latest [link for version 8](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Firmware/1351B08SIM7000E.rar) and the board was stuck in CFUN=7. I could not get AT+CFUN=1,1 to stick.  No error was thrown, but upon reboot it stayed in CFUN=7

So, I successfully downgraded to the next to the latest firmware, [link for version 7](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Firmware/1351B07SIM7000E.rar).  That was successful.
When I rebooted the shield it will not power on - it is blank. So, obviously I can not get to the diagnostic/debug serial port on the shield.
Not with just the 3.7V batter, nor with the Uno rev3 and the 3.7 V battery.

Any suggestions?
Is it now destined for the shield graveyard?
Thanks

botletics, 2022-01-13:
250mAH or 2500mAH?

johnedstone, 2022-01-13:
My apologies,  I corrected the typo above.  It is 3.7 2500 mAh from Adafruit [link to battery](https://www.adafruit.com/product/328)

johnedstone, 2022-01-13:
Updated issue with the correct information, sorry
2500 mAh

-------------------------
Ed Johnstone    Ô¿Ô¬
cell: 317.331.1848
email: ***@***.***
*'Discouragement is never from God, as it clouds faith and hope' - St
Ignatius*

*'Be the first to love' - Chiara Lubich*

botletics, 2022-01-13:
Interesting. I did notice one time when I was playing with updating the firmware on a shield, that the LEDs didn't turn on but I was still able to communicate with it through USB. I don't remember exactly what caused it, but I think redoing the firmware update solved it... But it looks like yours is just dead...

johnedstone, 2022-01-13:
Yes, I've kept looking for the serial port to show up (on both Windows and
Linux) .... and nothing, as I wondered if it was just an LED.  Thanks

-------------------------
Ed Johnstone    Ô¿Ô¬
cell: 317.331.1848
email: ***@***.***
*'Discouragement is never from God, as it clouds faith and hope' - St
Ignatius*

*'Be the first to love' - Chiara Lubich*