botletics/SIM7000-LTE-Shield #288: MQTT Example, keep resetting

dnwk, 2022-02-24:
I loaded MQTT example to my Arduino UNO and SIM7000C, it keeps resetting. I do have battery attached and LTE_demo runs ok.

** SIM7000 MQTT Example ***
20:39:54.689 -> Configuring to 9600 baud
20:39:54.792 -> Attempting to open comm with ATs
20:39:54.861 -> 	---> AT
20:39:54.861 -> 	<--- OK
20:39:54.895 -> 	---> ATE0
20:39:54.928 -> 	<--- OK
20:39:55.065 -> 	---> ATE0
20:39:55.065 -> 	<--- OK
20:39:55.232 -> 	---> AT+GMR
20:39:55.737 -> 	<--- Revision:1529B01SIM7000G
20:39:55.770 -> 
20:39:55.770 -> OK
20:39:55.770 -> 
20:39:55.770 -> 	---> AT+CPMS="SM","SM","SM"
20:39:55.840 -> 	<--- +CPMS: 0,30,0,30,0,30
20:39:55.874 -> FONA is OK
20:39:55.874 -> Found SIM7000
20:39:55.874 -> 	---> AT+GSN
20:39:55.907 -> 	<--- 86

It just keeps repeating above.

dnwk, 2022-02-24:
Now it is doing this 

20:57:08.013 -> 	---> AT+GSN
20:57:08.013 -> 	<--- 869[IMEI number]
20:57:08.047 -> Module IMEI: 869[IMEI number]
20:57:08.081 -> ⸮gO#⸮⸮߮Yᱷ⸮⸮⸮*** SIM7000 MQTT Example ***

I notice the gibberish output after IMEI. Baud rate issue?

dnwk, 2022-02-24:
em... I really didn't do anything now my Arduino can't communicate with the SIM7000. No response from AT command. USB direct connection to my computer still works. Did some components fail on the board?

botletics, 2022-02-24:
Does the battery have the correct polarity? Red wire should be on the left side of the connector when you plug it into the board.

dnwk, 2022-02-24:
Yes, I have confirmed battery connection is correct

botletics, 2022-02-24:
Please check the physical connection on pins 10-11 on the shield. If the green "PWR" LED on the board is staying on but you're not able to communicate with it, this may indicate an issue with the hardware connection being flaky.

closed 2022-03-03 by dnwk