botletics/SIM7000-LTE-Shield #290: No network connection ?

reddy9698, 2022-03-12:
Hi, I've been using the module for quite some days now and it was working all fine.
Suddenly since yesterday the module doesn't seem to get a network connection.
I use the same SIM on my phone and it works fine but on the module the status LED keeps blinking every 1s.
The same SIM also works fine on other boards.
I've even tried using other SIM cards but to no avail.
Even tried changing the antenna and still the same.

Is it a problem with the board? Please Help

I even tried the following AT commands :
AT+CPIN?
+CPIN : READY

AT+CSQ
+CSQ : 99,99

AT+CGREG?
+CGREG : 0,2

AT+COPS?
+COPS: 0

botletics, 2022-03-12:
So you mean the SIM card also works on other SIM7000 boards? It's likely a config issue. Please see the commands mentioned on [this wiki page] (https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network)

reddy9698, 2022-03-12:
YES, the same sim card works on other boards, but not on my board.
I tried the link that you provided. 
AT command gives the following response : 

AT+CREG?
+CREG=0,0

AT+CREG=?
+CREG=(0-2)

botletics, 2022-03-12:
Are you using a Botletics board?

reddy9698, 2022-03-12:
Yes

botletics, 2022-03-12:
Make sure you have a LiPo battery connected to it with proper polarity (red wire on left side of the connector when you plug the battery into the shield)

reddy9698, 2022-03-12:
Yes, That's all the same. As I mentioned earlier, I have been working on it for quite some time now. It was working fine yesterday and suddenly the network dropped and I haven't gotten any connection since then.

botletics, 2022-03-12:
Hmmm that's strange. You tried rebooting?

reddy9698, 2022-03-12:
I tried rebooting, tried upgrading the firmware, tried many things but nothing seems to work.

botletics, 2022-03-12:
Hmmm yea I'm not sure, sorry 😬. Can it still read the SIM card CCID?

botletics, 2022-03-12:
Also, to get a baseline test, make sure to load the LTE_Demo sketch into the Arduino and test it, not whatever code you had on it before. Or unplug it from the Arduino altogether and test via USB.

reddy9698, 2022-03-12:
No It returns an error for AT+CCID.
Yes I've tried both those methods.

botletics, 2022-03-12:
Hmm, maybe somehow the SIM card circuitry is fried. Sounds unusual, but not impossible I guess... 🤔

reddy9698, 2022-03-12:
Is there any way to check that?

thesoaringpotato, 2022-03-12:
Perhaps try issuing a hard reset, then launch the LTE_Demo sketch and specifically copy and paste the results of each of the following commands:

[C] Read the SIM CCID
[i] Read signal strength (RSSI)
[n] Get network status
[1] Get network connection info

reddy9698, 2022-03-12:
Tried that just now, Nothing's changed

SIM CCID : 

AT+CCID
ERROR

Signal Strength : 
AT+CSQ
+CSQ: 99,99

There's no further network status or connection info.