botletics/SIM7000-LTE-Shield #292: HTTPS - Library needs modifications to make it work.

richcj10, 2022-04-11:
I got the SIM7000A up and running over HTTP(s) last night. It was a bit PITA but was self explanatory once you follow the documentation to a T. 

I tested the code with pushsafer and pushover. Pushover requires HTTPs. 

The library I believe needs some modifications to make this work. 

`
        if (! sendCheckReply(F("AT+CIFSR"), ok_reply, 10000))
          return false;
`
This requests an IP in the "enableGPRS" function (after line 1632). I don't think this is strictly necessary. But speeds up the process. The device needs to have an IP. 

The other option that needs modification is the HTTP_ssl function.

`
boolean Adafruit_FONA::HTTP_ssl(boolean onoff) {
  sendCheckReply(F("AT+CSSLCFG=\"sslversion\",1,3"), ok_reply, 10000);
  return sendCheckReply(F("AT+SHSSL=1,\"\""), ok_reply,10000);
}
`

When you boot / setup the modem, you need to make sure you enable WirelessConnStatus:
`
  fona.enableGPRS(true);
  if (!fona.wirelessConnStatus()) {
    while (!fona.openWirelessConnection(true)) {
      Serial.println(F("Failed to enable connection, retrying..."));
      delay(2000); // Retry every 2s
    }
    Serial.println(F("Enabled data!"));
  }
  delay(100);
  fona.wirelessConnStatus();
`

Then we can send the data:
`
void Pushover(const char* Title, const char* Message){
  char URL[200];
  char BODY[200];
  sprintf(URL, "api.pushover.net");
  fona.HTTP_ssl(true);
  fona.HTTP_connect(URL);
  int SizeOfArray = sprintf(BODY, "token=YourTokenHere&user=YourUserIDHere&device=droid4&title=%s&message=%s",urlencode(Title).c_str(),urlencode(Message).c_str());
  Serial.print("Array Body = ");Serial.println(SizeOfArray);
  fona.HTTP_addHeader("Host","api.pushover.net",16);
  fona.HTTP_addHeader("Content-Type","application/x-www-form-urlencoded",33);
  //fona.HTTP_addHeader("Content-Length","140",3);
  fona.HTTP_POST("/1/messages.json",BODY,SizeOfArray);
}
`
The above function is good, it works. The provided function of post uses the wrong CMDs to setup the modem. The order is important. This should be easy to modify to make this work with any POST request. Enjoy. I am uploading  a full .ino file soon to show everything A->Z.

richcj10, 2022-04-11:
Here is the repo of the full piece of firmware:
https://github.com/richcj10/HTTPs-SIM7000A-LTE-Demo

Mark-Wills, 2022-04-11:
> Here is the repo of the full piece of firmware: https://github.com/richcj10/HTTPs-SIM7000A-LTE-Demo

Is the complete code (not including the Fona library) that is needed to run HTTPS posts/gets?

Thanks

richcj10, 2022-06-03:
I don't follow your question? As above, the function needs modified.

ZakiOro, 2022-06-04:
> I don't follow your question? As above, the function needs modified.

ah yes, after careful reading, yes it needs the certificate as second parameter

richcj10, 2022-06-05:
Yes, especially if length is wrong.

Nafana, 2022-08-10:
Hi @richcj10, I'm trying to figure out some issues with making a TLS connection on my SIM7000A. Any chance you can post the AT command log for when your module initializes and establishes a connection?

I've upgraded firmware (to 1351B04SIM7000A) and tried all sorts of suggestions (yours included), however, no succeess so far. Just looking to eliminate possible issues.
- I'm on the Telus' CAT-M1 network with shared Shared NAT’d IPs

ZakiOro, 2022-08-18:
> Hi @richcj10, I'm trying to figure out some issues with making a TLS connection on my SIM7000A. Any chance you can post the AT command log for when your module initializes and establishes a [...]

try to configure the SNI that is required if it's a shared ip, also make sure to load the correct certificate

botletics, 2022-09-08:
Hey guys, figured out how to connect with SSL without verifying certs! Please do the following:
- Set "SSL_FONA" to 1 in the .h file
- Download the latest .cpp file from GitHub
- Open the unedited LTE_Demo example sketch and change "http://dweet.io" to "https://dweet.io" on line 1035 (under the '2' option)
- Upload to the Arduino
- Run the 'G' command to enable data connection, then run '2' to send data to dweet.io using SSL. And voila! 😊

botletics, 2022-09-08:
Added this to the fona.HTTP_connect() function:

`if (SSL_FONA) {
    sendCheckReply(F("AT+CSSLCFG=\"sslversion\",1,3"), ok_reply);
    sendCheckReply(F("AT+SHSSL=1,\"\""), ok_reply, 10000);
  }`

scottdo10, 2022-10-21:
> Hey guys, figured out how to connect with SSL without verifying certs! Please do the following:
>
> * Set "SSL_FONA" to 1 in the .h file
> * Download the latest .cpp file from GitHub
> * Open the unedited [...]

Doesn't line 1035 only effect the 7070?  I've tried these changes suggested in the thread with a 7000A and still can't get https to work.

Testing with line 1064, http works but when changing it to https, it fails 
HTTP
	---> AT+HTTPPARA="URL","http://dweet.io/dweet/for/865235030717215?temp=765.06&batt=4065"
	<--- OK
	---> AT+HTTPACTION=0
	<--- OK
         HTTP status: 200

HTTPS
	---> AT+HTTPPARA="URL","https://dweet.io/dweet/for/865235030717215?temp=765.06&batt=4064"
	<--- OK
	---> AT+HTTPACTION=0
	<--- OK
         HTTP status: 603

richcj10, 2022-10-21:
What modem do you have and what is the full RX/TX log?

richcj10, 2022-10-26:
Glad you did! 
FYI: Some of my code is actually incorrect. I noticed the SIM7000 actually makes header info for you! ( I forget what, you can get this info by using a POST test place)  So some of my additional headder info caused problems with Pushover. Removing additional headder info fixed it.