botletics/SIM7000-LTE-Shield #293: DFRobot SIM7000 with botletics library GPRS and GPS wont work

danielkp1234, 2022-04-19:
Hello.
Firstly I know that this library isn't supposed to be used with the DFRobot sim7000 but I have been recommended by multiple people to use this library since apparently, the DFRobot library won't work and is unstable.
 
I have been trying to make a GPS tracker using a DFRobot sim7000 to try and send longitude and latitude to an MQTT server using a 3g/4g connection but I can’t seem to get it working with either library.

 I read a guide and came up with this code for submitting longitude and latitude to an MQTT server using cellular.
https://pastebin.com/pj1eaDzP

This is the output of the Serial Monitor
[image]

and this is the code that causes the errors. I have also tried commenting out the EnableGPS code to see if GPRS would work but that doesn’t work either.
[image]

wilba84, 2022-04-20:
Ive been playing with this all day.
The issue you are having is connecting the arduino and sim7000 via serial. You cannot use hardware serial as the arduino uno only has a single hardware serial port, which is already linked to the onboard usb converter for debugging.

For the last 6 hours ive been trying to use software serial but the the sim7000 now doesnt work at all, my sim7600 doesnt work with soft serial, but does with hw serial.

If possible, get your hands on a mega and use hw serial

danielkp1234, 2022-04-20:
> Ive been playing with this all day. The issue you are having is connecting the arduino and sim7000 via serial. You cannot use hardware serial as the arduino uno only has a single hardware serial [...]

Do you get GPS and cellular connection working with mega board?

@Mark-Wills I didn't really understand very much of what you wrote sorry but i am using a DFRobot SIM7000 shield stacked on top of an Arduino UNO

I tested everything with AT Command Tester (https://m2msupport.net/m2msupport/download-at-command-tester/) and both cellular connection was working and I could get an IP and GPS also works so the problem is library-related. If this library isn't going to work for me how would I go about debugging and finding out why the attacthService function in the DFRobot library doesn't work since the GPS features in the DFRobot library works 

Here is DFRobot SIM7000 library with the only change being made is changing the APN from cmnet to internet in attacthService void 
https://gist.github.com/danielkp1234/ab2ea90e5b47bd3070b7c0707d06bb61

danielkp1234, 2022-04-20:
> I have had cellular and gps working on the sim7000. Unfortunately My module died yesterday or I would send you some screenshots

Damn thats sucks. Do you know how i could start debugging the DFRobot SIM7000 libary since it kinda works its only the celluar part of the libary that has some kind of error since the same AT commands work when ran using AT command tester

wilba84, 2022-04-21:
The screenshot you sent above for the serial monitor, the sim7000 isnt actually replying, its a false flag, hence being detected as sim7500.
Are you plugging the usb directly into the sim7000 to use the AT Command tester?

danielkp1234, 2022-04-21:
> The screenshot you sent above for the serial monitor, the sim7000 isnt actually replying, its a false flag, hence being detected as sim7500.
> Are you plugging the usb directly into the sim7000 to use the AT Command tester?

Yes, i am

wilba84, 2022-04-21:
You need to establish a serial connection from UNO>SIM7000, either with software serial or hw serial. From memory if you put a blank sketch on the arduino and move the serial switch to 0/1, you should be able to plug usb into the arduino, open serial monitor and send AT commands

danielkp1234, 2022-04-21:
Well, if this wont work imma just going to try to find the issue in the attacthService function in the DFRobot libaray its properly some simple thing that is wrong. I just dont know how to debug it

DevinCarpenter, 2022-12-08:
@danielkp1234 any update on your issue?