botletics/SIM7000-LTE-Shield #298: Modem connected via USB is sometimes not responding…

ctschach, 2022-05-26:
Hi,
I have a Waveshare SIM7000G module connected via USB to a Raspberry Pi.

I frequently send AT commands for HTTP Posts to the modem on /dev/ttyUSB2.

That works fine for a while until the modem stopped to respond to AT commands.

Now I noticed, that I can connect to /dev/ttyUSB3 and send a single AT. Once I did this, the modem also responds to AT commands on  /dev/ttyUSB2 again.

Doesn‘t make a lot of sense to me. Any ideas?

thanks…

Mark-Wills, 2022-05-27:
Sounds a like firmware issue on the modem. In such a case, it is a good
idea to reset the modem (either by its reset pin, or by removing power and
re-applying power under software control from the raspberry pi).

ctschach, 2022-05-27:
> Sounds a like firmware issue on the modem. In such a case, it is a good idea to reset the modem (either by its reset pin, or by removing power and re-applying power under software control from the raspberry pi).
> […](#)

I do not have the reset pin connected as I only use The USB connection. What do you mean with „re-applying power“ on the Raspberry Pi? How do I control the power of a plugged USB device? Was not aware that this isnpossible.

BTW, I tried to use „AT+CFUN=6“ in /dev/ttyUSB3 and that do reset the modem. However, the problem of not responding reoccurs much faster.