botletics/SIM7000-LTE-Shield #317: LTE_Demo While statements blocking code

alejandrog1994, 2022-12-07:
Hello,

I'm working on a project which builds heavily on the LTE_Demo in order to send SMS when sensors on as esp32 detect certain conditions. These SMS functions are meant to complement the project, not be completely reliant on them. An issue I'm having is that if fona fails to initialize, it wont proceed with the rest of the sketch, or if I try to post data to dweet and it fails, it stops the entire program. Not asking to be spoonfed code, but I'm really struggling to find a solution that allows for failure of the FONA functions without stopping the rest of the program. I would very much appreciate advice on how to overcome this obstacle, Thank you

```
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x3F, 20, 4);
#include <Wire.h>
#include <OneWireNg.h>
#include <DallasTemperature.h>
#include <ezButton.h>
#include <Adafruit_ADS1X15.h>
#include <MultiMap.h>
#define ONE_WIRE_BUS 4
/***********************************GPRS/SMS STUFF ****************************************/
#include "Adafruit_FONA.h" // https://github.com/botletics/SIM7000-LTE-Shield/tree/master/Code
#include <HardwareSerial.h>
// For SIM7000 shield
#define FONA_PWRKEY 18
#define FONA_RST 5
#define FONA_TX 16 // ESP32 hardware serial RX2 (GPIO16)
#define FONA_RX 17 // ESP32 hardware serial TX2 (GPIO17)

#define SIMCOM_7000
#define PROTOCOL_HTTP_GET         // Generic
#define LED 2

HardwareSerial fonaSS(1);
Adafruit_FONA_LTE fona = Adafruit_FONA_LTE();

#define samplingRate 60000UL // The time in between posts, in millis
#define samplingRate2 3000UL // the time between sensor readouts
uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);
char imei[16] = {0}; // Use this for device ID
char replybuffer[255]; // Large buffer for replies
uint8_t type;
unsigned long counter = 0;
unsigned long counter2 = 0;
/**** BOOL VARIABLES FOR SMS ALERTS ****/
bool bat = false;
bool opened = false;
bool cold = false;

char volBuff[12];
char tempBuff[12];
char batBuff[12];

char URL[300];  // Make sure this is long enough for your request URL
char body[200]; // Only need this is you're doing an HTTP POST request

/**** FLOAT SWITCH ****/
const int  buttonPin = 33;
int buttonState = 0;         // current state of the button
int lastButtonState = 0;     // previous state of the button

const char * phone_number = "+10000000000"; // Include country code, only numbe
/*****                  SMS ALERTS                     ********/
const char * text_message1 = " mL FULL!"; // Change to suit your needs
const char * text_messageV = volBuff;

const char * text_message3 = " % BATTERY LOW!";
const char * text_messageB = batBuff;

const char * text_message2 = "*F FREEZING!"; // Change to suit your needs
const char * text_messageF = tempBuff;

volatile int count; //This integer needs to be set as volatile to ensure it updates correctly during the interrupt process.

float tLimit = 45.0; //the amount of liquid in mL that the pump is supposed to collect during pump cycle
int Threshold;       //the threshold for amount of liquid collected before pump returns to OffState
[... 806 lines cut ...]
  //  digitalWrite(PWRKEY, LOW);
  //  delay(600); // Minimum of 64ms to turn on and 500ms to turn off for FONA 3G. Check spec sheet for other types
  //  delay(1300); // Minimum of 1.2s for SIM7000
  //  digitalWrite(PWRKEY, HIGH);

  // Shut down the MCU to save power
#ifndef samplingRate
  Serial.println(F("Shutting down..."));
  delay(5); // This is just to read the response of the last AT command before shutting down
  MCU_powerDown(); // You could also write your own function to make it sleep for a certain duration instead
#else
  //  // The following lines are for if you want to periodically post data (like GPS tracker)
  //  Serial.print(F("Waiting for ")); Serial.print(samplingRate); Serial.println(F(" seconds\r\n"));
  //  delay(samplingRate * 1000UL); // Delay

  // Only run the initialization again if the module was powered off
  // since it resets back to 115200 baud instead of 4800.
#ifdef turnOffShield
  fona.powerOn(FONA_PWRKEY); // Powers on the module if it was off previously
  moduleSetup();
#endif

#endif
}
```

botletics, 2022-12-07:
That's a lot of code to look through, but the first thing that comes to mind is to delete this line:

`while(1); // Don't proceed if it couldn't find the device`

in the `moduleSetup()` function.

willburk97, 2022-12-07:
I think I revised it in my code to just try for a set number of millis and then move on.  I'm using an older version of the repository and my code's pretty messy, sorry, but it's here:
https://github.com/willburk97/CH4100_SIM7000

-Will

alejandrog1994, 2023-01-18:
> 

Thanks for the help, I did what you did by doing the following   
```
// Turn on GPRS
  while (!fona.enableGPRS(true) &&(millis() < 30000)) {
    Serial.println(F("Failed to enable GPRS, retrying..."));
    delay(2000); // Retry every 2s
  }
  Serial.println(F("Enabled GPRS!"));
#endif
#endif
```
I added this same millis delay to the following functions 

```
void FullSMS () {
  char message3[20];

  strcpy(message3, volBuff);
  strcat(message3, text_message1);

  pinMode(FONA_RST, OUTPUT);
  digitalWrite(FONA_RST, HIGH); // Default state
  pinMode(FONA_PWRKEY, OUTPUT);
  pinMode(buttonPin, INPUT_PULLUP);
  powerOn(true); // Power on the module
  moduleSetup(); // Establish first-time serial comm and print IMEI

  fona.setNetworkSettings(F("hologram")); // For Hologram SIM card, change appropriately
  while (!netStatus()&&(millis() < 30000)) {
    Serial.println(F("Failed to connect to cell network, retrying..."));
    delay(2000); // Retry every 2s
  }
  Serial.println(F("Connected to cell network!"));

  // Send a text to your phone!
  if (!fona.sendSMS(phone_number, message3)) {
    Serial.println(F("Failed to send text!"));
  }
  else {
    Serial.println(F("Sent text alert!"));
  }
}

void ColdSMS () {
  char message[20];

  strcpy(message, tempBuff);
  strcat(message, text_message2);
  pinMode(FONA_RST, OUTPUT);
  digitalWrite(FONA_RST, HIGH); // Default state
  pinMode(FONA_PWRKEY, OUTPUT);

  powerOn(true); // Power on the module
  moduleSetup(); // Establish first-time serial comm and print IMEI

  fona.setNetworkSettings(F("hologram")); // For Hologram SIM card, change appropriately
  while (!netStatus()&&(millis() < 30000)) {
    Serial.println(F("Failed to connect to cell network, retrying..."));
    delay(2000); // Retry every 2s
  }
  Serial.println(F("Connected to cell network!"));

  // Send a text to your phone!
  if (!fona.sendSMS(phone_number, message)) {
    Serial.println(F("Failed to send text!"));
  }
  else {
    Serial.println(F("Sent text alert!"));
[... 183 lines cut ...]
    pumpOff();
    lcd.setCursor(0, 0);
    lcd.print("FULL          ");
  }

  if (tempF < tLimit && cold == false) {
    ColdSMS ();
    cold = true;
  }
  else if (tempF > tLimit + 2.0 && cold == true) {
    cold = false;
  }
  if (percent < bLimit && bat == false) {
    batSMS ();
    bat = true;
  }
  /***************************SMS ALERTS END***************************************************/
  /**************************** DATA POST TO DWEET *********************************************/
  if (millis() - counter > samplingRate) {
    PostData ();
    counter = millis ();
  }
}
```
please advise