botletics/SIM7000-LTE-Shield #323: Couldn't find fona

Jen-n14, 2023-05-10:
Hi!
I would like to ask for help about my project. I am making a flood monitoring system with sms notification usinh arduino uno r3, gsm 900a module and water level sensor. Below is the wiring of my system.
I tried uploading the code and it was successful yet my serial monitor says couldn't find fona.
May I know what can I do to solve the issue?

Your help will be greatly appreciated💗

Code:
//Tech Trends Shameer
//Flood Alert Call and SMS

#include <SoftwareSerial.h>
#include "Adafruit_FONA.h"

#define FONA_RX 2
#define FONA_TX 3
#define FONA_RST 4

int resval = 0; // holds the value
int respin = A3; // Water Level Sensor Pin

#define FONA_RI_INTERRUPT 0
SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);

Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

char PHONE_1[21] = ""; // Enter your Mobile Number

char tempalert[141]= "Flood Alert" ;
void setup()
{
// pinMode(gas_sensor_pin,INPUT);
Serial.begin(115200);
// dht.begin();
Serial.println(F("Initializing....(May take 3 seconds)"));
delay(5000);
fonaSS.begin(9600); // if you're using software serial
if (! fona.begin(fonaSS)) { // can also try fona.begin(Serial1)
Serial.println(F("Couldn't find FONA"));
while (1);
}

fona.print ("AT+CSMP=17,167,0,0\r");
Serial.println(F("FONA is OK"));
//fona.sendSMS(PHONE_1, welcomemessage);
// pinMode(gas_sensor_pin, INPUT);

}

void loop(){

resval = analogRead(respin);

Serial.print("Water Level:");
Serial.println(resval);

if(resval>200 )
{
Serial.println("Flood Alert");
make_multi_call();
send_multi_sms();
}

}

void send_multi_sms()
{
if(PHONE_1 != ""){
Serial.print("Phone 1: ");
fona.sendSMS(PHONE_1,tempalert);
delay(20000);
}

}
void make_multi_call()
{
if(PHONE_1 != ""){
Serial.print("Phone 1: ");
make_call(PHONE_1);
delay(20000);
}

}

void make_call(String phone)
{
Serial.println("calling....");
fona.println("ATD"+phone+";");
delay(20000); //20 sec delay
fona.println("ATH");
delay(1000); //1 sec delay
}

Wiring :
S -> A3

-> 5v
-> GND
Rx -> tx of arduino
Tx -> rx of arduino
VCC -> 5v
GND -> GND
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mark-Wills, 2023-05-11:
The problem is almost certainly the amount of current required to properly
drive the fona board. Depending on the modem type, it can need a LOT of
current to properly operate the modem, especially when the modem is first
registering on the network, and when it is transmitting. For example, the
SIM800L modem needs up to 2 AMPS (in very short bursts) to operate reliably.

The 5V supply that you are giving it cannot supply enough current. My
suggestion - just to get things working - would be to use a bench power
supply set to the correct voltage. Connect the GND of the bench power
supply and the Arduino together so that they are properly referenced to
each other.

If you don't have access to a bench power supply, get one of these buck
converters: https://www.aliexpress.com/item/1005002772498639.html

If the fona board needs 5V, then feed the buck converter with say 9 to 12V
and it will easily supply the fona board. I have used these exact boards
with a SIM800L prototype (needs up to 2 amps!) and it works 100% reliably.
Use good cables - not those cheap dupont cables that we all use on
breadboards - they don't have enough copper in them and cannot supply the
current - they act like resistors at high current.

Hope this helps.

Mark

pmethods, 2023-08-13:
My only input is that my botletics gear worked exactly as advertised

-methods