botletics/SIM7000-LTE-Shield #325: HTTP-601 - Arduino Nano

arch-linux, 2023-08-06:
I have taken pieces of the [Bioletics SIM7000 Demo Code](https://raw.githubusercontent.com/botletics/Botletics-SIM7000/main/examples/LTE_Demo/LTE_Demo.ino) in an attempt to get temperature data from my arduino to a Node.JS Server with automation.

When running the bioletics demo code, the following commands function perfectly, I run the following order.

G (enable GMRS data) - w (get) -> http://23.94.248.142:3000/test ( this test appears on my webserver console just fine)

The following code takes the setup of the normal bioletics file, combines it with a SHT31 temperature probe and attempts to transmit it.

`
#include "BotleticsSIM7000.h" // https://github.com/botletics/Botletics-SIM7000/tree/main/src
#include <Adafruit_SHT31.h>
#include <Arduino.h>
#include <Wire.h>
#include <SoftwareSerial.h>

#define SIMCOM_7000

#define PWRKEY 6
#define RST 7
#define TX PD7 // Microcontroller RX
#define RX PD6 // Microcontroller TX
char replybuffer[255];

#include <SoftwareSerial.h>
SoftwareSerial modemSS = SoftwareSerial(TX, RX);
Adafruit_SHT31 sht31 = Adafruit_SHT31();

SoftwareSerial *modemSerial = &modemSS;

#ifdef SIMCOM_2G
  Botletics_modem modem = Botletics_modem(RST);
  
// Use this one for 3G modules
#elif defined(SIMCOM_3G)
  Botletics_modem_3G modem = Botletics_modem_3G(RST);

#elif defined(SIMCOM_7000) || defined(SIMCOM_7070) || defined(SIMCOM_7500) || defined(SIMCOM_7600)
  Botletics_modem_LTE modem = Botletics_modem_LTE();
#endif

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);
uint8_t type;
char imei[16] = {0}; // MUST use a 16 character buffer for IMEI!

void setup() {

  Serial.begin(9600);
  Serial.println(F("Modem basic test"));
  Serial.println(F("Initializing....(May take several seconds)"));
  modemSS.begin(115200); 
  Serial.println(F("Configuring to 9600 baud"));
  modemSS.println("AT+IPR=9600"); // Set baud rate
  delay(100); // Short pause to let the command run
  modemSS.begin(9600);
  if (! modem.begin(modemSS)) {
    Serial.println(F("Couldn't find modem"));
    while (1); // Don't proceed if it couldn't find the device
  }

 

  type = modem.type();
  Serial.println(F("Modem is OK"));
  Serial.print(F("Found "));
  switch (type) {
    case SIM800L:
      Serial.println(F("SIM800L")); break;
    case SIM800H:
      Serial.println(F("SIM800H")); break;
    case SIM808_V1:
      Serial.println(F("SIM808 (v1)")); break;
    case SIM808_V2:
      Serial.println(F("SIM808 (v2)")); break;
    case SIM5320A:
      Serial.println(F("SIM5320A (American)")); break;
    case SIM5320E:
      Serial.println(F("SIM5320E (European)")); break;
    case SIM7000:
      Serial.println(F("SIM7000")); break;
    case SIM7070:
      Serial.println(F("SIM7070")); break;
    case SIM7500:
      Serial.println(F("SIM7500")); break;
    case SIM7600:
      Serial.println(F("SIM7600")); break;
    default:
      Serial.println(F("???")); break;
  }

  uint8_t imeiLen = modem.getIMEI(imei);
  if (imeiLen > 0) {
    Serial.print("Module IMEI: "); Serial.println(imei);
  }
  modem.setFunctionality(1); 
  modem.setNetworkSettings(F("hologram"));

 
  if(!modem.GPRSstate()){
    if(modem.GPRSstate()){
      Serial.println("GPRS ENABLED OK");
    } else {
      Serial.println("GPRS NOT ENABLED.");
      modem.enableGPRS(true);
      // one last final check
      delay(10000);
      if(modem.GPRSstate()){
        Serial.println("GPRS ENABLED OK");
      } else {
        Serial.println("GPRS did not enable after final check");
      }
    }
  } else {
    Serial.println("GMRS ENABLED OK");
  }

  //START TEMPERATURE PROBE
  if (! sht31.begin(0x44)) {   // Set to 0x45 for alternate i2c addr
    Serial.println("Couldn't find SHT31");
    while (1) delay(1);
  }

}

int cTOf(float celsius) {
  return int(celsius * 9.0 / 5.0 + 32.0);
}

void flushSerial() {
  while (Serial.available())
    Serial.read();
}

void getURL(int temp, float humidity){

// read website URL
        uint16_t statuscode;
        int16_t length;
        char url[80];

        flushSerial();
        sprintf(url,"http://23.94.248.142:3000/");
        delay(10);
        sprintf(url, "temperature/%d/%f", temp,humidity);
        delay(100);
        Serial.println(F("****"));
        if (!modem.HTTP_GET_start(url, &statuscode, (uint16_t *)&length)) {
          Serial.println("Failed!");

        } else {
while (length > 0) {
          while (modem.available()) {
            char c = modem.read();

            // Serial.write is too slow, we'll write directly to Serial register!
#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega168__)
            Serial.println("HERE");
            loop_until_bit_is_set(UCSR0A, UDRE0); /* Wait until data register empty. */
            UDR0 = c;
#else
            Serial.write(c);
#endif
            length--;
        }
        Serial.println(F("\n****"));
        modem.HTTP_GET_end();

      }
        }
        

}

void loop(){
         delay(5000);
        uint8_t n = modem.getNetworkStatus();
        if(n == 5){
           Serial.println(F("Registered roaming"));
         }

        delay(2000);
        float t = sht31.readTemperature();
        float h = sht31.readHumidity();
        char URL[150];
        char body[100];
        char tempBuff[16];
        int tempConverted = cTOf(t);
        getURL(tempConverted,h);
        
        delay(500000);
}
`
The result is unfortunately a 601 error.

  ---> AT+HTTPTERM <--- OK ---> AT+HTTPINIT <--- OK ---> AT+HTTPPARA="CID" <--- OK ---> AT+HTTPPARA="UA" <--- OK ---> AT+HTTPPARA="URL" <--- OK ---> AT+HTTPACTION=0 <--- OK Status: 601 Len: 0 ---> AT+HTTPREAD <--- OK Failed!

I was expecting the typical response from my server of "URL logged to the console.".

Unfortunately I get a 601.

What am I doing wrong?

Thank you for any help you can provide.

arch-linux, 2023-08-06:
I am aware of the sprintf issue, that has been fixed. same result unfortunately

Mark-Wills, 2023-08-06:
My suspicion is the port - you are using port 3000. Please try accessing
the front page of any website (e.g. google.com or microsoft.com on) on port
80 first. Narrow the problem down. If you can access google on port 80,
then try accessing your server on port 80. If you can access your server on
port 80, but not on port 3000 then it is a firewall issue. You have opened
port 3000 on the receiving server, right?

arch-linux, 2023-08-06:
@Mark-Wills 

Changing my code to request only http://google.com results in the same failed error. 

```
#include "BotleticsSIM7000.h" // https://github.com/botletics/Botletics-SIM7000/tree/main/src

#include <Adafruit_SHT31.h>

#include <Arduino.h>

#include <Wire.h>

#include <SoftwareSerial.h>

#define SIMCOM_7000

#define PWRKEY 6
#define RST 7
#define TX PD7 // Microcontroller RX
#define RX PD6 // Microcontroller TX
char replybuffer[255];

#include <SoftwareSerial.h>

SoftwareSerial modemSS = SoftwareSerial(TX, RX);
Adafruit_SHT31 sht31 = Adafruit_SHT31();

SoftwareSerial * modemSerial = & modemSS;

#ifdef SIMCOM_2G
Botletics_modem modem = Botletics_modem(RST);

// Use this one for 3G modules
#elif defined(SIMCOM_3G)
Botletics_modem_3G modem = Botletics_modem_3G(RST);

#elif defined(SIMCOM_7000) || defined(SIMCOM_7070) || defined(SIMCOM_7500) || defined(SIMCOM_7600)
Botletics_modem_LTE modem = Botletics_modem_LTE();
#endif

uint8_t readline(char * buff, uint8_t maxbuff, uint16_t timeout = 0);
uint8_t type;
char imei[16] = {
  0
}; // MUST use a 16 character buffer for IMEI!

void setup() {

  Serial.begin(9600);
  Serial.println(F("Modem basic test"));
  Serial.println(F("Initializing....(May take several seconds)"));
  modemSS.begin(115200);
  Serial.println(F("Configuring to 9600 baud"));
  modemSS.println("AT+IPR=9600"); // Set baud rate
  delay(100); // Short pause to let the command run
  modemSS.begin(9600);
  if (!modem.begin(modemSS)) {
    Serial.println(F("Couldn't find modem"));
    while (1); // Don't proceed if it couldn't find the device
  }

  type = modem.type();
  Serial.println(F("Modem is OK"));
  Serial.print(F("Found "));
  switch (type) {
  case SIM800L:
    Serial.println(F("SIM800L"));
    break;
  case SIM800H:
    Serial.println(F("SIM800H"));
    break;
  case SIM808_V1:
    Serial.println(F("SIM808 (v1)"));
    break;
  case SIM808_V2:
    Serial.println(F("SIM808 (v2)"));
    break;
  case SIM5320A:
    Serial.println(F("SIM5320A (American)"));
    break;
  case SIM5320E:
    Serial.println(F("SIM5320E (European)"));
    break;
  case SIM7000:
    Serial.println(F("SIM7000"));
    break;
  case SIM7070:
    Serial.println(F("SIM7070"));
    break;
  case SIM7500:
    Serial.println(F("SIM7500"));
    break;
  case SIM7600:
    Serial.println(F("SIM7600"));
    break;
  default:
    Serial.println(F("???"));
    break;
  }

  uint8_t imeiLen = modem.getIMEI(imei);
  if (imeiLen > 0) {
    Serial.print("Module IMEI: ");
    Serial.println(imei);
  }
  modem.setFunctionality(1);
  modem.setNetworkSettings(F("hologram"));

  if (!modem.GPRSstate()) {
    if (modem.GPRSstate()) {
      Serial.println("GPRS ENABLED OK");
    } else {
      Serial.println("GPRS NOT ENABLED.");
      modem.enableGPRS(true);
      // one last final check
      delay(10000);
      if (modem.GPRSstate()) {
        Serial.println("GPRS ENABLED OK");
      } else {
        Serial.println("GPRS did not enable after final check");
      }
    }
  } else {
    Serial.println("GPRS ENABLED OK");
  }

  //START TEMPERATURE PROBE
  if (!sht31.begin(0x44)) { // Set to 0x45 for alternate i2c addr
    Serial.println("Couldn't find SHT31");
    while (1) delay(1);
  }

  delay(5000);
  uint8_t n = modem.getNetworkStatus();
  if (n == 5) {
    Serial.println(F("Registered roaming"));
  }

}

int cTOf(float celsius) {
  return int(celsius * 9.0 / 5.0 + 32.0);
}

void flushSerial() {
  while (Serial.available())
    Serial.read();
}

void getURL(int temp, float humidity) {

  // read website URL
  uint16_t statuscode;
  int16_t length;
  char *url = "http://google.com";

  flushSerial();
  Serial.println(F("****"));

        if (!modem.HTTP_GET_start(url, &statuscode, (uint16_t *)&length)) {
          Serial.println("Failed!");
        }
        while (length > 0) {
          while (modem.available()) {
            char c = modem.read();

            // Serial.write is too slow, we'll write directly to Serial register!
#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega168__)
            loop_until_bit_is_set(UCSR0A, UDRE0); /* Wait until data register empty. */
            UDR0 = c;
#else
            Serial.write(c);
#endif
            length--;
            if (! length) break;
          }
        }
        Serial.println(F("\n****"));
        modem.HTTP_GET_end();
      }

void loop() {

  delay(2000);
  float t = sht31.readTemperature();
  float h = sht31.readHumidity();
  char URL[150];
  char body[100];
  char tempBuff[16];
  int tempConverted = cTOf(t);
  getURL(tempConverted, h);

  delay(500000);
}
```

In my original question, I poised that the default code provided by botletics works just fine, even with my URL. 
When I copied the exact code, minus the readline inputs, this results in a 601. 

```
	<--- OK
	---> AT+HTTPPARA="CID"
	<--- OK
	---> AT+HTTPPARA="UA"
	<--- OK
	---> AT+HTTPPARA="URL"
	<--- OK
	---> AT+HTTPACTION=0
	<--- OK
Status: 601
Len: 0
	---> AT+HTTPREAD
	<--- OK
Failed!

****
	---> AT+HTTPTERM
	<--- OK
```

It is unclear what/if I am missing to make this issue occur.

botletics, 2023-08-09:
It would help for you to post more of your serial monitor to troubleshoot. Code 601 usually means your device isn't actually connected to the internet, usually by an unsuccessful "enableGPRS(true)"